AEFeinstein/Super-2024-Swadge-FW #20: Memory/Save/Data Manager

Brycey92, 2023-08-08:
**Summary**
_What is the feature all about? Is it a game, a utility, an enhancement, or something else?_
Memory manager from the 2023 1.1 update, but with per-mode namespaces

**Technical Spec**
_How will this be implemented?_
_What peripherals will be used, and how?_
In 2023, all accesses to NVS used the same namespace. With different namespaces, we can create a polished save manager, and can more easily reset single modes' data when they misbehave.

AEFeinstein, 2023-08-08:
> when they misbehave

Pfft, they won't

Brycey92, 2023-10-12:
It was decided this mode is not needed, and individual modes should make their own data reset/management as-needed case-by-case.

closed 2023-10-12 by Brycey92

AEFeinstein, 2023-10-12:
Yep, my game as a 'reset data' option on the menu, and a second menu level as confirmation :)