AEFeinstein/Super-2024-Swadge-FW #26: Sequencer/tracker

DeadlySprinklez, 2023-08-08:
**Summary**
A simple tracker for music creation, live, on the Swadge. BPM modification, two-channel support for chords, and a loop toggle are included.

**Technical Spec**

- Main screen: D-Pad moves around the timeline and up/down in scale. Touchpad sliding for quick scrubbing. A to place a left-channel note, and B to place a right-channel note. Only one of each note can be on the same beat at the same time - attempting to place another note will move the note instead. Start to play/pause, and Select to toggle into a track modification menu.
- Track modification menu: D-Pad Up and Down to select options, D-Pad Left and Right to modify BPM if selected. Start to save, Select to cancel.

**UI**
Deep blue to dark blue gradient background with music sheet-like white lines drawn across the screen. A yellow reticle is visible on the track, shaped like two arrow brackets (< >) which can place notes when the corresponding button is pressed. The notes are a red bubble with a white L, and a light blue bubble with a black R, respectively.

**Mockups**
(To be added)

**How to Test**
If notes are placed and the play button is pressed, the timeline should manifest a playhead and it should scroll from the beginning of the track to the end of the track, playing the notes to the beat when the playhead arrives at them.

AEFeinstein, 2023-09-16:
@DeadlySprinklez any anything on this?

AEFeinstein, 2023-10-21:
Putting this off until next year

DeadlySprinklez, 2024-05-05:
Picking this up again!

AEFeinstein, 2024-05-05:
Woo!

AEFeinstein, 2024-09-13:
@DeadlySprinklez let me know on Discord that the week between 9/10 and 9/17 is for evaluation, and we'll check after then if this is feasible, or gets shelved / handed off.

closed 2024-10-28 by AEFeinstein