AEFeinstein/Super-2024-Swadge-FW #36: Port Gamepad

Brycey92, 2023-08-14:
**Summary**
_What is the feature all about? Is it a game, a utility, an enhancement, or something else?_
Gamepad mode from 2023, now in IDF 5.0

**Technical Spec**
_How will this be implemented?_
_What peripherals will be used, and how?_
Digital touch buttons should be removed, and replaced with a virtual analog stick mapped to the new touchpad. 2024 hardware will be needed for this.

**UI**
_What are the display screens & menus (if applicable)?_
_What user inputs are there for each screen?_
Same UI as last year, but with changes for aforementioned touchpad redesign. We can also add a deadzone setting if we find it natural to return the finger on the touchpad to a "home" position in the center, rather than lifting the finger.

**Mockups**
_Paste any mockups of screens or anything else here._
Sample analog stick visualization with deadzone:
[image]

**How to Test**
How can we verify the feature is working?
Test on a Switch and a PC

Brycey92, 2023-08-17:
The bulk of the mode is currently ported, except for aforementioned UI/control changes and the gamepad support in the tinyusb component.

No compiling or testing has been done, since the tinyusb changes are required.

AEFeinstein, 2023-08-17:
If you want to make the UI look more Metroid-Primey, and less gamecube controllery, that'd be cool too. Maybe some color swaps, maybe some HUD decor?

Brycey92, 2023-08-27:
Currently blocked by missing driver support. Looking for helper(s) experienced with modifying components.

Will pick up UI and touchpad changes once driver compiles.

AEFeinstein, 2023-08-27:
Driver support isn't missing, but I agree that it isn't working as expected. When loading demo mode, my Swadge _does_ enumerate as a USB Gamepad, but it isn't sending button presses as expected. When I ported everything, I used `TUD_HID_REPORT_DESC_GAMEPAD`, but it looks like that was replaced in our source by `TUD_HID_REPORT_DESC_GAMEPAD_SWADGE`. I suspect something about that is screwy. Paging @cnlohr 
* Docu: https://docs.espressif.com/projects/esp-idf/en/v5.1/esp32s2/api-reference/peripherals/usb_device.html
* Example: https://github.com/espressif/esp-idf/tree/v5.1/examples/peripherals/usb/device/tusb_hid

AEFeinstein, 2023-08-27:
@Brycey92 in the meantime you can try checking out 2bf02263f92fe2401efbe2f98b22edb7c51a8df0, the rev before advanced USB changes, and see if Demo mode's USB works for you there.

Brycey92, 2023-08-28:
I ran a fullclean, `make`, and `idf.py build flash` on https://github.com/AEFeinstein/Swadge-IDF-5.0/commit/2bf02263f92fe2401efbe2f98b22edb7c51a8df0 and the hardware becomes unresponsive when powered on. No LEDs light up, the TFT remains off, and the Swadge cannot be rebooted into the bootloader.

A forced programming mode boot with the D-pad UP held allows reflashing to known working firmware from `main` branch, though repeating steps to clean and build the suggested commit result in the same issues.

AEFeinstein, 2023-08-28:
dang. Lets go further back, does it work in 30823ed? If so, this may be a job for `git bisect`

AEFeinstein, 2023-08-28:
Also, fwiw, this was originally developed on a squarewavebird

Brycey92, 2023-10-12:
This now compiles, connects to PC, and reports buttons, thanks to @AEFeinstein.

We now have to
- update its UI for the new touchpad (@Brycey92)
- add Nintendo Switch compat (@JVeg199X, @cnlohr)
- add settings to change touchpad reporting style btwn 4/8/4+ctr/8+ctr/analog/etc (@dylwhich? @cnlohr? @JVeg199X? me? one of us should claim this if we have time.)

AEFeinstein, 2023-10-21:
@Brycey92 motion support and PC/Switch compat are in `main`, so I think this is all in your court for UI and maybe some config if you want.

closed 2023-11-30 by AEFeinstein