AEFeinstein/Super-2024-Swadge-FW #43: Emulator Extensions

dylwhich, 2023-08-21:
**Summary**
_What is the feature all about? Is it a game, a utility, an enhancement, or something else?_

This feature is an enhancement to the swadge emulator to add optional "extensions" which would make it easier to include additional functionality without needing to rewrite core emulation functionality which could potentially introduce bugs or inconsistencies.

Each extension would be able to read and/or manipulate the emulator state before and after each frame, handle or modify keyboard/mouse inputs, send inputs to the swadge, and request a window "pane" on any edge of the window where they can draw any UI elements. This should be sufficient to implement anything we would reasonably want to.

**Technical Spec**
_How will this be implemented?_
_What peripherals will be used, and how?_

An emulator extension would work similarly to a swadge mode, in that it is defined by a struct with several callbacks. Unlike a swadge mode, any number of extensions could be active at once, and all extensions would need to be registered in some central "loader" structure. Command-line args could enable or disable extensions, and extensions could use several callbacks to interact with the emulator.

The general structure of an extension:
* Init callback: Looks at command-line args and returns whether or not the extension is enabled. Should also request any window panes needed for UI components and allocate any memory, etc.
* Keyboard callback: Gets called any time a key is pressed. The callback may consume the key event, or replace it with a new key.
* Mouse Motion callback: Gets called any time the mouse is moved. The callback may consume the event.
* Mouse button callback: Gets called any time a mouse button is pressed. The callback may consume the event.
* Pre-frame callback: Gets called immediately before the swadge's main loop resumes after yielding
* Render callback: Gets called just before the framebuffer is drawn, with the window/pane dimensions passed in
* Post-frame callback: Gets called immediately after the swadge's main loop yields back to the system

The emulator flow with extensions:
1. Command-line arguments are parsed into a struct
2. **Extensions are initialized based on the command-line args**
3. Emulator starts up
4. Swadge starts up
5. Swadge main loop runs
6. **Post-frame callback**
7. Swadge inputs and timers are called
8. Render swadge display
9. **Render callback**
10. **Pre-frame callback**
11. Repeat at step 5

**UI**
_What are the display screens & menus (if applicable)?_
_What user inputs are there for each screen?_

The current emulator would need to be changed from the current hardcoded-ish layout, to a more dynamic layout with, otherwise, the same behavior. The LEDs would be allocated panes just like extensions, and the panes would be re-laid-out every time the window changes size.

**Mockups**
_Paste any mockups of screens or anything else here._

**How to Test**
How can we verify the feature is working?

AEFeinstein, 2023-08-22:
Do you have plans for any specific extensions? FWIW, I didn't pull in last year's CLI stuff, so off the cuff we're missing
* Fuzzing
* Starting at a specific mode
* Locking to a specific mode
* Fullscreen

Of those, fuzzing seems extension-able

dylwhich, 2023-08-22:
Yeah, that sounds about right. Starting in / locking modes might be doable as an extension as-is, not sure though. I think aside from those there could be:
*  Virtual touchpad
* LEDs, mainly for layout consistency
* Remapping keys for selfish/lazy reasons
* Recording/playing back inputs? would be very cool and neat for testing
* Native screenshots?? Combined with recorded inputs, could be useful
* Something that shows your inputs on-screen

AEFeinstein, 2023-08-22:
Those all are reasonable features. I wonder if shimming everything in to the extension framework is the correct approach. Like, the features could also be built individually. It also feels a bit odd to me to have a virtual touchpad be an extension while all other emulated hardware components are built-in.

dylwhich, 2023-08-23:
Yeah, I guess the way I think of it is the emulator should probably natively implement all the peripherals that have a direct analogue already on a computer (sound/mic, nvs, display, keyboard input...). But even then, anything that goes beyond the actual swadge's normal capability would just be functions exposed for an extension to use. Then, anything that would require hardware (accelerometer, touchpad, leds, rumble motor...) would have the swadge-side functionality mostly stubbed out, and then the actual implementation is handled by an extension that writes the simulated values based on input and/or displays something based on the swadge data.

Honestly, you could make a case for the display and keyboard input being extensions as well. Then the core emulator would provide the framebuffer access and a function to inject button events, and the extensions would handle just drawing the framebuffer and turning the key events into buttons. I guess you could think of it more as a "driver" than an extension where all the peripherals are concerned.

At a certain point, it's really just a matter of organization, so we don't have to go crazy making everything fit into an extension. I think it's mostly just much easier to have anything that needs to take input or draw UI elements do that as an extension, even if the main functionality is still part of the core emulator.

AEFeinstein, 2023-08-23:
I agree this is a matter of organization. My overall philosophy was to have as little emulator-specific code as possible. That meant `main()` does a little setup and calls `app_main()`, the firmware's main function which does not return. `taskYIELD()`, called from `app_main()`, does emulator loop stuff. It also meant `emulator/src/components/` re-implements everything in `components/`, using the same headers, keeping a similar folder structure. That means something like `getTouchCentroid()` was implemented in `hdw-btn.c`.

I'm still not grasping the upside of an extension framework versus implementing features directly, but I do like moving the drawing logic from `taskYIELD()` to the emulated components.

Either way, we should use [`argp`](https://www.gnu.org/software/libc/manual/html_node/Argp.html) for argument parsing.

dylwhich, 2023-08-23:
> I'm still not grasping the upside of an extension framework versus implementing features directly, but I do like moving the drawing logic from taskYIELD() to the emulated components.

Yeah, I think the drawing logic being moved out of `taskYIELD()` is probably the main benefit, since that's where all the really annoying complexity starts to come in. And if we want to be able to show and hide other UI bits the same way as the LEDs it gets crazy fast, so we'd want something to handle the layout for that.

I think the other main benefit is just in having a consistent place to plug things in. Where all of the fuzzing and mode-locking stuff came together last year it made it tricky to work in that area sometimes -- I think doing things this with extensions makes it easier to keep track of what's "vanilla" emulator stuff and what's extra features changing the functionality. If there's something we want to do outside of what the hooks allow (e.g. disabling the exit mode button sequence directly) maybe it's better to add an explicit flag to do that, and then just let extensions control it.

Or maybe I'm way overthinking it... 

> Either way, we should use [argp](https://www.gnu.org/software/libc/manual/html_node/Argp.html) for argument parsing.

Ah, yes! I couldn't remember if there was a cross-platform compatibility issue last year with using whichever arg parsing library. We should indeed use that.

AEFeinstein, 2023-08-23:
Your head's definitely in the right place, but I think you're overthinking it by a little bit. I like organization and fully support
* Moving drawing logic out of `taskYIELD()` (though they still will get called through `taskYIELD()`'s call stack
* Organizing features cleanly into individual files (`fuzzing.c`, `modeLock.c`, etc.)
* Enabling or disabling feature flags in a consistent manner
* Using `argp` (it does nicer things with help strings and args than `getopt`)

The only thing I remain hesitant about is trying to force all of this into a swadge-mode like struct stuffed with function pointers. Swadge modes are all similar enough to impose object-oriented-like inheritance and are developed by a relatively large number of folk. Extensions would be more varied and likely developed by you and maybe me, so the overhead & rigidity of a struct doesn't seem worth it.

closed 2023-08-31 by dylwhich