AEFeinstein/Super-2024-Swadge-FW #45: Untitled Marble-Matching Game

dylwhich, 2023-08-25:
**Summary**
This is a game where each level has a track in which a certain number of colorful marbles slowly move along, towards the end. The object is to eliminate all these marbles before any of them reach the end. If any marbles reach the end, the player loses a life and restarts.

The player controls an object that shoots marbles and can be rotated a full 360 degrees. The player may eliminate marbles in the track by shooting a marble between marbles on the track to create a group of 3 or more marbles of the same color. The color of the player's marble is randomized and unknown until it appears. When a group of marbles is eliminated, the marbles that were on either side of the group will move towards each other until they touch. If, again, there are 3 or more marbles of the same color touching, that group will also be eliminated. This allows for chain reactions which will give the player lots of bonus points.

There will be at least some hand-made levels, and if they can be made fun enough, there will be endless procedurally-generated levels.

Some marbles in the track may contain power-ups, and could either act as a regular marble of a specific color, or some sort of wildcard. The powerup would be activated or earned when the marble containing it is eliminated. Some powerup marbles could turn into a regular marble if they aren't eliminated after a certain amount of time. Some ideas for powerups:

* Bomb-marble that removes a large number of nearby marbles when eliminated
* M-m-m-multi-ball
* Store a marble for later
* See what the next marble will be
* Wildcard marble
* Bouncy marbles that can skip a row of the track

This game will also feature some tower-defense aspects. The user will be able to buy/unlock certain items and place them into the level, possibly by hand or in a set location. These would include:
* Mini-turret that automatically and slowly shoots marbles of a single color (plus upgraded versions)
* Some sort of road-block thing that stops the marbles from moving for a while

Other random ideas:
* Levels where you control multiple marble shooters simultaneously
* Levels with multiple tracks

**Technical Spec**
Each level will be defined by a structure that has:
* The location of the marble shooter
* A path for the marbles to follow (probably some kind of spine)
* Level settings such as marble colors, speed, powerups available, number of marbles, etc.

The main gameplay loop will move the marbles along the path in groups, joining them when the touch so that all marbles that are touching move consistently. It will also handle movement and collision detecting for any marbles that were shot. It will also do some elastic collision physics stuff when groups of marbles collide.

The touchpad will be used, iPod-style, for controlling the rotation of the marble shooter.

**UI**
The mode would be broken up into a main menu with some sort of level select, and the game itself.

**Mockups**
_Paste any mockups of screens or anything else here._

**How to Test**
How can we verify the feature is working?

AEFeinstein, 2023-08-25:
You're describing Zuma, right?

dylwhich, 2023-08-26:
> You're describing Zuma, right?

Yeah pretty much

AEFeinstein, 2023-10-21:
Putting this one off for next year.

closed 2023-10-21 by AEFeinstein