AEFeinstein/Super-2024-Swadge-FW #52: Namespaced NVS

Brycey92, 2023-08-27:
**Summary**
_What is the feature all about? Is it a game, a utility, an enhancement, or something else?_
Using mode-specific namespaces will help keep NVS more tidy and free up additional characters in variable names since mode prefixes will be OBE.

**Technical Spec**
_How will this be implemented?_
_What peripherals will be used, and how?_
The NVS driver will need to be modified to expose the namespace parameters to user code.

**UI**
_What are the display screens & menus (if applicable)?_
_What user inputs are there for each screen?_
N/A

**Mockups**
_Paste any mockups of screens or anything else here._
N/A

**How to Test**
How can we verify the feature is working?
After implementing namespaces in all modes that use NVS, test those modes on both emulator and hardware for proper saving/loading.

Brycey92, 2023-08-27:
To do: figure out if we want to keep the old namespace-less API in addition to the new one, or change the API and all calls to it.

Brycey92, 2023-10-12:
@dylwhich you mentioned you had already started implementing this in your branch, so I assigned you to the issue.

AEFeinstein, 2023-10-21:
Added in #93

closed 2023-10-21 by AEFeinstein