AEFeinstein/Super-2024-Swadge-FW #67: Crash on start button in emulator while in mode

MrTroyD, 2023-09-15:
**Description**
When using the emulator and a game mode. Pressing the start button closes out the emulator. 

**Expected behavior**
The quick settings menu should appear.

**Reproduction Steps**
Open the swadge emulator. From main menu start demo by pression A. When the next menu screen begins, press Start. Notice that the desktop app closes.

**Logs**
==30542==ERROR: AddressSanitizer: negative-size-param: (size=-2)

**Affected Version**
 - Linux
 - main
 - git hash: ce9723d

**Additional Notes**
swadge2024.c :438 needs something to make sure that numPx is never below zero. This will fix the problem. I added 
`
if (numPx < 0) numPx = abs(numPx); 
`
To my code and that fixed the problem. I'm pretty sure there's a better way to do that.

AEFeinstein, 2023-09-15:
@dylwhich FYI

dylwhich, 2023-09-15:
Ah, I think this might be what `fix-quick-settings-crash` was meant to fix, but I was never able to reproduce it again so I couldn't tell if it actually worked.

dylwhich, 2023-09-16:
Okay, so this is actually happening because `timePausePressed` is, somehow, in the future. It happens quite often, but only occasionally does it manage to get far enough into the future for the final result to be greater than zero.

closed 2023-09-16 by AEFeinstein