AEFeinstein/Super-2024-Swadge-FW #73: Makefile target to generate .vscode/c_cpp_properties.json

Brycey92, 2023-09-16:
**Summary**
_What is the feature all about? Is it a game, a utility, an enhancement, or something else?_
This will generate a file that helps VS Code find elements of the IDF toolchain, and helps both with and without the extension.

**Technical Spec**
_How will this be implemented?_
_What peripherals will be used, and how?_
My current idea is to implement this with a Makefile target that takes a `.json.in` file and replaces some text with the path to the user's home folder, producing the `.json` file. Using `${userHome}` directly in the `.json` file on a Windows machine results in Unix-like paths (`/home/gelak/`) as opposed to Windows-like paths (`C:\Users\gelak\`). This doesn't work, as the redirection seems to be limited to some other software (IDF itself, MSYS2 terminals, something else) and does not extend to VS Code's reading of the paths.

Bonus points if we can get the exact version numbers to be found in the filesystem and text-replaced in this file without needing to update the file manually after an IDF update.

**How to Test**
How can we verify the feature is working?
The generated `.json` file on @AEFeinstein's computer should look like this:
```
{
    "configurations": [
        {
            "name": "esp32s2",
            "includePath": [
                "${workspaceFolder}/main/**",
                "${workspaceFolder}/components/**",
                "C:/Users/gelak/esp/esp-idf/components/**"
            ],
            "compilerPath": "C:/Users/gelak/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc.exe",
            "cStandard": "c17",
            "cppStandard": "c++17",
            "intelliSenseMode": "gcc-x86",
            "compileCommands": "${workspaceFolder}/build/compile_commands.json"
        }
    ],
    "version": 4
}
```

AEFeinstein, 2023-11-01:
```bash
PROPERTIES_FILE=./.vscode/c_cpp_properties.json

gen-vs-properties:
	@echo "{" >> $(PROPERTIES_FILE)
	@echo "    \"configurations\": [" >> $(PROPERTIES_FILE)
	@echo "        {" >> $(PROPERTIES_FILE)
	@echo "            \"name\": \"esp32s2\"," >> $(PROPERTIES_FILE)
	@echo "            \"includePath\": [" >> $(PROPERTIES_FILE)
	@echo "                \"\$${workspaceFolder}/main/**\"," >> $(PROPERTIES_FILE)
	@echo "                \"\$${workspaceFolder}/components/**\"" >> $(PROPERTIES_FILE)
	@echo "            ]," >> $(PROPERTIES_FILE)
	@echo "            \"defines\": []," >> $(PROPERTIES_FILE)
	@echo "            \"compilerPath\": \"$(shell which xtensa-esp32s2-elf-gcc)\"," >> $(PROPERTIES_FILE)
	@echo "            \"compileCommands\": \"\$${workspaceFolder}/build/compile_commands.json\"," >> $(PROPERTIES_FILE)
	@echo "            \"cStandard\": \"c17\"," >> $(PROPERTIES_FILE)
	@echo "            \"cppStandard\": \"c++17\"," >> $(PROPERTIES_FILE)
	@echo "            \"intelliSenseMode\": \"gcc-x86\"," >> $(PROPERTIES_FILE)
	@echo "            \"configurationProvider\": \"ms-vscode.makefile-tools\"" >> $(PROPERTIES_FILE)
	@echo "        }" >> $(PROPERTIES_FILE)
	@echo "    ]," >> $(PROPERTIES_FILE)
	@echo "    \"version\": 4" >> $(PROPERTIES_FILE)
	@echo "}" >> $(PROPERTIES_FILE)
```

AEFeinstein, 2024-05-05:
At some point this got checked in: https://github.com/AEFeinstein/Super-2024-Swadge-FW/blob/main/.vscode/c_cpp_properties.json

The Linux config does require `/usr/bin/gcc`, which I think is reasonable. The ESP-IDF does have a kinda ugly path to the compiler, but it _should_ be consistent if the setup instructions are followed. There isn't a Windows config, but if someone wants it, we can add it.

closed 2024-05-05 by AEFeinstein