AEFeinstein/Super-2024-Swadge-FW #90: Improve Factory Test motion detection

Brycey92, 2023-10-12:
**Summary**
_What is the feature all about? Is it a game, a utility, an enhancement, or something else?_
It would be very nice for the Factory Test to verify more than just "each x/y/z value is > 500." Since I don't know how the accel works, Adam suggested you work on this @cnlohr.

**Technical Spec**
_How will this be implemented?_
_What peripherals will be used, and how?_

**UI**
_What are the display screens & menus (if applicable)?_
_What user inputs are there for each screen?_

**Mockups**
_Paste any mockups of screens or anything else here._

**How to Test**
How can we verify the feature is working?

AEFeinstein, 2023-10-12:
Related to #55

AEFeinstein, 2023-10-21:
@cnlohr is there anything you still want to do here or is this closeable?

AEFeinstein, 2023-11-01:
@cnlohr I'm assuming this is all done, so I'm closing it. Re-open if you have something in mind.

closed 2023-11-01 by AEFeinstein