AEFeinstein/Super-2024-Swadge-FW #97: Dice Roller / Board Game Utility

dylwhich, 2023-10-19:
**Summary**
This mode is a utility that includes a dice roller, a timer, and possibly other useful features like counters or tracking player order.

There was a dice roller utility last year, which allowed selecting up to 6 dice of the same type, with either 2, 4, 6, 8, 10, 12, 20, or 100-sided dice, and showed 6 rolls of history with the sum of each roll. So at a minimum, we should be able to support that.

List of feature ideas, regardless of difficulty to implement:
* Using multiple types of dice at once
* Groups of dice that are rolled at once but scored separately (think Risk)
* Presets for popular games that configure/disable the timer, set up number and type of dice, and possibly game-specific interpretations of the dice (again, think Risk, Monopoly)
* User-configurable and saveable presets
* Broadcast info to other swadges that can view and/or roll dice when it's their turn

**Technical Spec**
Dice rolls: This will use the IMU and tiny physics engine to simulate and render 3D virtual dice shaking inside a box based on how the user shakes the swadge.

Timer: This will be a countdown timer, using the swadge's time APIs. The timer can be configured to a specific time and easily paused or reset back to the starting time. The timer may play a tune and/or flash the LEDs when it runs out.

**UI**
The starting screen will probably be a menu with some quick options - Dice, Timer, Presets, Custom.

Then, the primary screen would be shown once one of the presets or dice/timer-only are opened. It would show a large timer display if the timer is enabled, and a 2D display that shows the last dice rolled.

To roll dice, the user would either just shake the swadge, or press/hold some button and shake the swadge. Or maybe just press a button to automatically roll the dice for lazy people. While that happens, some or all of the screen may be replaced with the physics simulation of the dice until the dice have settled. Ideally, the state of the dice should automatically be detected once they settle. Then, either automatically or once the user continues, the primary screen returns and the latest roll results are shown there.

For other screen elements such as counters, moving left/right on the D-pad moves between elements, and up/down changes the values of those elements. A and B will depend on the type of element selected. Context-dependent controls will be shown at the bottom of the screen if there ends up being room. Any other inputs needed could be done with the touchpad (for example, the timer could use iPod spins to set a longer more quickly/precisely).

**Mockups**
_Paste any mockups of screens or anything else here._

**How to Test**
How can we verify the feature is working?

AEFeinstein, 2023-10-19:
@cnlohr FYI

Brycey92, 2023-11-18:
Are we planning on implementing this within 5 days? Or porting the old one as-is?

dylwhich, 2023-11-18:
There's a sort of working prototype in `dice-roller-thing` that maybe someone could get working. I won't have time, but if someone wants to try and tweak the physics numbers and add some polish it could still happen.

dylwhich, 2024-05-10:
Gonna close this for now unless someone else wants to take it on for this year.

closed 2024-05-10 by dylwhich