AEFeinstein/Super-2024-Swadge-FW #126: Unify framebuffer location

cnlohr, 2023-11-18:
This is a long-term concern.  Nothing needed immediately.

It is a not-insignificant perf hit to call `getPxTftFramebuffer` or to reach into and get a global pixel register value.  However, using file-static variables are fast. 

It would be really good if we could avoid the function call, or global read.

I was seeing ~10% perf boosts on the flight sim by making these changes.

AEFeinstein, 2023-11-18:
The framebuffer is malloc'd rather than being file static. I think if you want to static it, we'll also need to steal some memory from the heap in the linker.
```c
if (NULL == pixels)
{
    pixels = (paletteColor_t*)malloc(sizeof(paletteColor_t) * TFT_HEIGHT * TFT_WIDTH);
}
```

cnlohr, 2023-11-19:
No need to static it or not - its getting access to the pointer that's the hard/slow part.  Making a project-global variable takes an inordinate amount of time to load at the beginning of the function.

AEFeinstein, 2023-11-19:
OK. If it's really that easy and impacts flightsim, do it in that branch?

cnlohr, 2023-11-19:
Good to have a discussion on - I am doing it where it counts most for flightsim, but could improve wsg timing as well.

AEFeinstein, 2023-11-19:
Is there any downside?

AEFeinstein, 2024-07-21:
Closing until we need the perf boost.

closed 2024-07-21 by AEFeinstein