AEFeinstein/Super-2024-Swadge-FW #155: .mid to .sng converter occasionally produces bad output

Brycey92, 2023-11-22:
These midis produce identical .sng files (only 1 included) that are much shorter than all other .sng files. They don't play in firmware or the emulator, and seem to act like 0-note songs.
[jukebox.zip](https://github.com/AEFeinstein/Swadge-IDF-5.0/files/13445073/jukebox.zip)

AEFeinstein, 2023-11-23:
When I've run into MIDIs that don't convert right, I've loaded them up in https://signal.vercel.app/?lang=en or https://www.audacityteam.org/ and re-saved them. That's helped.

Brycey92, 2023-11-29:
Just updated the issue description with an excerpt from a DM between FuryfulFawful and I.

AEFeinstein, 2023-11-29:
FWIW I used https://github.com/Gegel85/midi_parser to parse the MIDIs.. There's zero documentation.

AEFeinstein, 2024-05-05:
Will keep this in mind when expanding on MIDI processing this year.

AEFeinstein, 2024-06-10:
Closing b/c it looks like we're going to use MIDI as-is this year.

closed 2024-06-10 by AEFeinstein