AEFeinstein/Super-2024-Swadge-FW #156: Emulator (and maybe therefore firmware) has memory leaks

Brycey92, 2023-11-23:
```
vanillyneko@vanillyneko-Zephyrus-M-GM501GS:~/esp/Swadge-IDF-5.0$ ./swadge_emulator --touch --fuzz-motion
Extension touch initializing... enabled!
Extension leds initializing... enabled!
Extension fuzzer initializing... 
Fuzzing:
 - [ ] Buttons
 - [ ] Touch
 - [X] Motion
enabled!
Extension keymap initializing... disabled!
Extension modes initializing... disabled!
Extension replay initializing... disabled!
initUsb is UNIMPLEMENTED
checkAndInstallCrashwrap is UNIMPLEMENTED
Pulse: from: (null)/(null) (cnfgsound) / 8000x(2,1) (256)
Pulse initialized.
initTemperatureSensor is UNIMPLEMENTED
deinitTemperatureSensor is UNIMPLEMENTED
deinitUsb is UNIMPLEMENTED
HandleDestroy is UNIMPLEMENTED

=================================================================
==16253==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 592 byte(s) in 1 object(s) allocated from:
    #0 0x55a9c44c17c7 in calloc (/home/vanillyneko/esp/Swadge-IDF-5.0/swadge_emulator+0x1197c7)
    #1 0x7f2e44139c00  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xb3c00)

Indirect leak of 1264 byte(s) in 2 object(s) allocated from:
    #0 0x55a9c44c17c7 in calloc (/home/vanillyneko/esp/Swadge-IDF-5.0/swadge_emulator+0x1197c7)
    #1 0x7f2e44e6b4b1  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xde54b1)
    #2 0x7f2e44e6deea  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xde7eea)

Indirect leak of 632 byte(s) in 1 object(s) allocated from:
    #0 0x55a9c44c17c7 in calloc (/home/vanillyneko/esp/Swadge-IDF-5.0/swadge_emulator+0x1197c7)
    #1 0x7f2e44e6b4b1  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xde54b1)
    #2 0x7f2e44e6e5e0  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xde85e0)

Indirect leak of 216 byte(s) in 1 object(s) allocated from:
    #0 0x55a9c44c17c7 in calloc (/home/vanillyneko/esp/Swadge-IDF-5.0/swadge_emulator+0x1197c7)
    #1 0x7f2e44e5a183  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xdd4183)
    #2 0x7f2e44e5cf7d  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xdd6f7d)
    #3 0x7f2e44e6e128  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xde8128)

Indirect leak of 216 byte(s) in 1 object(s) allocated from:
    #0 0x55a9c44c17c7 in calloc (/home/vanillyneko/esp/Swadge-IDF-5.0/swadge_emulator+0x1197c7)
    #1 0x7f2e44e5a183  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xdd4183)
    #2 0x7f2e44e5cf7d  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xdd6f7d)
    #3 0x7f2e44e6e67e  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xde867e)

Indirect leak of 96 byte(s) in 2 object(s) allocated from:
    #0 0x55a9c44c19a8 in __interceptor_realloc (/home/vanillyneko/esp/Swadge-IDF-5.0/swadge_emulator+0x1199a8)
    #1 0x7f2e44e71d79  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xdebd79)
    #2 0x7f2e44e606fe  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xdda6fe)

Indirect leak of 48 byte(s) in 1 object(s) allocated from:
    #0 0x55a9c44c17c7 in calloc (/home/vanillyneko/esp/Swadge-IDF-5.0/swadge_emulator+0x1197c7)
    #1 0x7f2e44e61990  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xddb990)

Indirect leak of 40 byte(s) in 1 object(s) allocated from:
    #0 0x55a9c44c17c7 in calloc (/home/vanillyneko/esp/Swadge-IDF-5.0/swadge_emulator+0x1197c7)
    #1 0x7f2e44e60544  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xdda544)

Indirect leak of 12 byte(s) in 1 object(s) allocated from:
    #0 0x55a9c44c15f7 in __interceptor_malloc (/home/vanillyneko/esp/Swadge-IDF-5.0/swadge_emulator+0x1195f7)
    #1 0x7f2e44e6bfa6  (/usr/lib/x86_64-linux-gnu/dri/iris_dri.so+0xde5fa6)

SUMMARY: AddressSanitizer: 3116 byte(s) leaked in 11 allocation(s).
```

AEFeinstein, 2023-11-23:
I don't think this is an issue in firmware. None of those leaks reference specific files, and the emulator does have some leaky system stuff (at least that it complains about on WSL)

ADC-AFeinstein, 2024-03-01:
Also all of these reference `iris_dri.so` which appears to be part of the `[libgl1-mesa-dri](https://packages.ubuntu.com/focal/libgl1-mesa-dri)` package. If there's a leak here, it's coming from [rawdraw](https://github.com/cntools/rawdraw).

AEFeinstein, 2024-05-04:
I don't know if this is WSL or not, but I do know WSL leaks memory. Unless asan can point at something specific in the swadge code, and not `iris_dri.so`, there's nothing to do here.

closed 2024-05-04 by AEFeinstein