AEFeinstein/Super-2024-Swadge-FW #200: FUTURE: Increase sensitivity and/or power

cnlohr, 2024-01-21:
Just a TINY bit, like ONE notch.  It's just toooooo restrictive it seems.

AEFeinstein, 2024-01-21:
Do you mean like lighter touches should be registered?

AEFeinstein, 2024-07-21:
Will revisit when there are hotdog prototypes

cnlohr, 2024-07-22:
As a note for the future, the lines that matter are:

https://github.com/AEFeinstein/Super-2024-Swadge-FW/blob/main/components/hdw-esp-now/hdw-esp-now.c#L315-L318
```c
    // Appears to set gain "offset" like what it reports as gain?  Does not actually impact real gain.
    // But when paired with the second write command, it seems to have the intended impact.
    // This number is in ~1/2dB.  So 20 accounts for a 10dB muting; 25, 12.5dB; 30, 15dB
    const int igi_reduction = 20;
```

and 

https://github.com/AEFeinstein/Super-2024-Swadge-FW/blob/main/components/hdw-esp-now/hdw-esp-now.c#L200

```c
    wifi_country_t usa = {.cc = "USA", .schan = 1, .nchan = 11, .max_tx_power = 84, .policy = WIFI_COUNTRY_POLICY_AUTO};
```

cnlohr, 2024-07-22:
Wait, it looks like we are commanding max power??? I thought we were muting.  I wonder if there is muting code elsewhere I am unaware of.  Our comments mention `wifi_set_user_fixed_rate` - perhaps it was a previous version?

AEFeinstein, 2024-07-22:
I lost the tread on wifi a while ago, but I did try to turn the power up all the way when doing the initial IDF setup.

cnlohr, 2024-07-22:
aaahh https://github.com/AEFeinstein/Super-2024-Swadge-FW/blob/main/sdkconfig#L826-L828

```ini
CONFIG_ESP_PHY_MAX_WIFI_TX_POWER=20
CONFIG_ESP_PHY_MAX_TX_POWER=20
CONFIG_ESP_PHY_REDUCE_TX_POWER=y
CONFIG_ESP_PHY_RF_CAL_PARTIAL=y
```
Ok, we will have to investigate.