AEFeinstein/Super-2024-Swadge-FW #202: Problem with building emulator / firmware

MasterTimeThief, 2024-01-24:
**Description**
I cannot build the emulator or the firmware. The issue has hit me two seperate times, both of which I will describe here.

**Reproduction Steps**

1. The compilation would throw a bunch of warnings from the project, before treating one or more of them as a fatal error and stopping the build. I *was* able to fix this by going into the makefile and removing `-Werror=all` from the `CFLAGS_WARNINGS` variable.

[image]

2. After I fixed the previous issue, I tried building the emulator again and ran into a different problem. At first, it gave me an error that said `x86_64-w64-mingw32-gcc.exe` couldn't be found, so I tried reinstalling msys2 based on the guide. That led me to my current issue, which now gives me no actual errors. It just seems to stop after the first gcc command, and I don't see what's breaking it.

[image]

AEFeinstein, 2024-01-24:
What do you get when you run `x86_64-w64-mingw32-gcc.exe --version`? For reference, my output is below. If you get something different, that'd be a good lead. Also make sure the order of paths in your PATH variable matches the guide (specifically the python and msys ones)
```
x86_64-w64-mingw32-gcc.exe (Rev3, Built by MSYS2 project) 13.2.0
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
I don't see the error in your second screenshot either. Maybe try running `make clean` and `make all` as separate steps? Just for kicks, you can try running `make all` a second time too.

As a sidebar, it's better to suppress a specific warning with something like `-Wno-error=stringop-overflow` rather than turn them all off with `-Werror=all`

MasterTimeThief, 2024-01-24:
My output for gcc is the same as yours, and my path setup is exactly as you have it in the guide too. And yeah, I don't see an error either, that's where I'm stuck. It just shows the first command and then stops. running `make clean` and `make all` seperately did the same thing.

[image]

My best guess is it's not making the .o file, but I don't know why or how that broke.

AEFeinstein, 2024-01-25:
You can check if it's building that object file. It would be generated at `./emulator/obj/emulator/src/emu_main.o`. I suspect you're right that it isn't.

It's strange that your compiler and path would match, but still have this issue. You can try running the compile command outside the makefile with this long line. If there is an error message that make is hiding, maybe this will expose it.
```bash
x86_64-w64-mingw32-gcc.exe -c -g -static-libgcc -static-libstdc++ -fdiagnostics-color=always -ffunction-sections -fdata-sections -gdwarf-4 -ggdb -O2 -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -finline-functions -std=gnu17 -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -Wno-error=unused-but-set-variable -Wno-old-style-declaration -Wundef -Wformat=2 -Winvalid-pch -Wlogical-op -Wmissing-format-attribute -Wmissing-include-dirs -Wpointer-arith -Wunused-local-typedefs -Wuninitialized -Wshadow -Wredundant-decls -Wjump-misses-init -Wswitch -Wcast-align -Wformat-nonliteral -Wno-switch-default -Wunused -Wunused-macros -Wmissing-declarations -Wmissing-prototypes -Wcast-qual -Wno-switch -Wunused-result   -DCONFIG_IDF_TARGET_ESP32S2=y  -DSOC_RMT_CHANNELS_PER_GROUP=4  -DSOC_TOUCH_SENSOR_NUM=15  -DSOC_ULP_SUPPORTED=y  -DSOC_PM_SUPPORT_EXT_WAKEUP=y  -DSOC_GPIO_SUPPORT_SLP_SWITCH=y  -DSOC_TIMER_GROUP_TIMERS_PER_GROUP=2  -DSOC_TIMER_GROUPS=2  -DSOC_I2C_NUM=2  -DSOC_I2C_SUPPORT_SLAVE=y  -DSOC_LEDC_CHANNEL_NUM=8  -DSOC_UART_NUM=2  -DSOC_ADC_DIGI_RESULT_BYTES=2  -DCONFIG_ESP_TIMER_SUPPORTS_ISR_DISPATCH_METHOD=0  -DCONFIG_LOG_MAXIMUM_LEVEL=3  -DCONFIG_GC9307_240x280=y  -DCONFIG_TFT_MAX_BRIGHTNESS=200  -DCONFIG_TFT_MIN_BRIGHTNESS=10  -DCONFIG_NUM_LEDS=8  -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1  -D_GNU_SOURCE  -DIDF_VER="v5.1.1"  -DESP_PLATFORM  -D_POSIX_READER_WRITER_LOCKS  -DCFG_TUSB_MCU=OPT_MCU_ESP32S2  -DEMULATOR=1  -DGIT_SHA1=\"8c48c63\"  -DHAS_XINERAMA=1  -DFULL_SCREEN_STEAL_FOCUS=1  -Iemulator  -Iemulator/idf-inc  -Iemulator/idf-inc/class  -Iemulator/idf-inc/class/hid  -Iemulator/idf-inc/driver  -Iemulator/idf-inc/freertos  -Iemulator/idf-inc/hal  -Iemulator/idf-inc/rom  -Iemulator/idf-inc/rom/usb  -Iemulator/idf-inc/soc  -Iemulator/obj  -Iemulator/obj/emulator  -Iemulator/obj/emulator/src  -Iemulator/obj/emulator/src/components  -Iemulator/obj/emulator/src/components/crashwrap  -Iemulator/obj/emulator/src/components/hdw-battmon  -Iemulator/obj/emulator/src/components/hdw-btn  -Iemulator/obj/emulator/src/components/hdw-bzr  -Iemulator/obj/emulator/src/components/hdw-esp-now  -Iemulator/obj/emulator/src/components/hdw-imu  -Iemulator/obj/emulator/src/components/hdw-led  -Iemulator/obj/emulator/src/components/hdw-mic  -Iemulator/obj/emulator/src/components/hdw-nvs  -Iemulator/obj/emulator/src/components/hdw-spiffs  -Iemulator/obj/emulator/src/components/hdw-temperature  -Iemulator/obj/emulator/src/components/hdw-tft  -Iemulator/obj/emulator/src/components/hdw-usb  -Iemulator/obj/emulator/src/extensions  -Iemulator/obj/emulator/src/extensions/fuzzer  -Iemulator/obj/emulator/src/extensions/keymap  -Iemulator/obj/emulator/src/extensions/leds  -Iemulator/obj/emulator/src/extensions/modes  -Iemulator/obj/emulator/src/extensions/replay  -Iemulator/obj/emulator/src/extensions/touch  -Iemulator/obj/emulator/src/idf  -Iemulator/obj/emulator/src/sound  -Iemulator/obj/main  -Iemulator/obj/main/asset_loaders  -Iemulator/obj/main/asset_loaders/common  -Iemulator/obj/main/colorchord  -Iemulator/obj/main/display  -Iemulator/obj/main/menu  -Iemulator/obj/main/modes  -Iemulator/obj/main/modes/accelTest  -Iemulator/obj/main/modes/breakout  -Iemulator/obj/main/modes/colorchord  -Iemulator/obj/main/modes/credits  -Iemulator/obj/main/modes/dance  -Iemulator/obj/main/modes/demo  -Iemulator/obj/main/modes/factoryTest  -Iemulator/obj/main/modes/flight  -Iemulator/obj/main/modes/gamepad  -Iemulator/obj/main/modes/jukebox  -Iemulator/obj/main/modes/lumberjack  -Iemulator/obj/main/modes/mainMenu  -Iemulator/obj/main/modes/mfpaint  -Iemulator/obj/main/modes/platformer  -Iemulator/obj/main/modes/pushy  -Iemulator/obj/main/modes/quickSettings  -Iemulator/obj/main/modes/ray  -Iemulator/obj/main/modes/ray/enemies  -Iemulator/obj/main/modes/slideWhistle  -Iemulator/obj/main/modes/timer  -Iemulator/obj/main/modes/touchTest  -Iemulator/obj/main/modes/tunernome  -Iemulator/obj/main/utils  -Iemulator/src  -Iemulator/src/components  -Iemulator/src/components/crashwrap  -Iemulator/src/components/hdw-battmon  -Iemulator/src/components/hdw-btn  -Iemulator/src/components/hdw-bzr  -Iemulator/src/components/hdw-esp-now  -Iemulator/src/components/hdw-imu  -Iemulator/src/components/hdw-led  -Iemulator/src/components/hdw-mic  -Iemulator/src/components/hdw-nvs  -Iemulator/src/components/hdw-spiffs  -Iemulator/src/components/hdw-temperature  -Iemulator/src/components/hdw-tft  -Iemulator/src/components/hdw-usb  -Iemulator/src/extensions  -Iemulator/src/extensions/fuzzer  -Iemulator/src/extensions/keymap  -Iemulator/src/extensions/leds  -Iemulator/src/extensions/modes  -Iemulator/src/extensions/replay  -Iemulator/src/extensions/touch  -Iemulator/src/idf  -Iemulator/src/sound  -Imain  -Imain/asset_loaders  -Imain/asset_loaders/common  -Imain/colorchord  -Imain/display  -Imain/menu  -Imain/modes  -Imain/modes/accelTest  -Imain/modes/breakout  -Imain/modes/colorchord  -Imain/modes/credits  -Imain/modes/dance  -Imain/ [... cut ...] nernome  -Imain/utils  -Icomponents/crashwrap/include  -Icomponents/hdw-battmon/include  -Icomponents/hdw-btn/include  -Icomponents/hdw-bzr/include  -Icomponents/hdw-esp-now/include  -Icomponents/hdw-imu/include  -Icomponents/hdw-led/include  -Icomponents/hdw-mic/include  -Icomponents/hdw-nvs/include  -Icomponents/hdw-spiffs/include  -Icomponents/hdw-temperature/include  -Icomponents/hdw-tft/include  -Icomponents/hdw-usb/include emulator/src/emu_main.c -o emulator/obj/emulator/src/emu_main.o
```

MasterTimeThief, 2024-01-27:
It wasn't making the object file, even with the seperate command. However, I decided to try and update my entire msys2 install, and that seems to have fixed the issue. Took me a bit to realize because I had been able to compile the build once, but I may have updated `x86_64-w64-mingw32-gcc.exe` after that, and the mismatch of versions between that and msys2 as a whole might have been what broke stuff.

closed 2024-01-27 by MasterTimeThief

AEFeinstein, 2024-01-27:
Huh, glad you sorted it out. Happy hacking!