AEFeinstein/Super-2024-Swadge-FW #217: Swadge Zine

AEFeinstein, 2024-05-04:
**Summary**
Wouldn't it be neat to have a little comic zine on each Swadge, just like Tea Room has?

dylwhich, 2024-05-04:
Sounds like a good use case for resurrecting the markdown renderer! And I think the markdown parsing could actually be moved to be part of asset processing instead of doing it at runtime, so we'd just load up a static markdown tree and the accompanying text blob

AEFeinstein, 2024-05-05:
Oh, I figured this would be more of an image sideshow

dylwhich, 2024-05-10:
Ahh, I totally missed the "comic" part, woops. One of these days I'll find a problem that can be solved with markdown!!

AEFeinstein, 2024-05-10:
There was some chatter about a dating sim or text adventure or something, which would be ripe for MD!

AEFeinstein, 2024-09-13:
Artist wrangling is hard, so I'm gonna shelve this for now.

closed 2024-09-13 by AEFeinstein