AEFeinstein/Super-2024-Swadge-FW #221: Big Bug (title pending)

DebrisHauler, 2024-05-06:
**Summary**
You will control Ovo Garbotnik as he bounces off the tiles of the level to destroy them. Maybe carve out tunnels underground like in dig dug with some monsters or goodies to go after, or just destroy all the tiles like in breakout. I'm not really sure yet, but it will develop into one of those.

**Technical Spec**
Right now it implements directional controls. I would eventually like to add in support for touch pad and optionally accelerometer to control Ovo.

**UI**
Pause menu, and gameplay. Keeping it simple hopefully.

**Mockups**
None yet

**How to Test**
Play it, see if the physics feel right.

DebrisHauler, 2024-05-20:
https://github.com/AEFeinstein/Super-2024-Swadge-FW/assets/6497989/a480e639-f38a-4b2b-a682-0f4cdc8c33d9

DebrisHauler, 2024-05-20:
https://github.com/AEFeinstein/Super-2024-Swadge-FW/assets/6497989/85d123fa-2a40-4591-ba9f-6d1667f82686

DebrisHauler, 2024-05-20:
I need an artist's help for a tileable cave background png that is 255x255 resolution.

DebrisHauler, 2024-05-20:
Design change to the control scheme: why not just go full helldivers and hold down the 'a' button to type in a stratagem sequence on the d-pad. That leaves 'b' button available for shooting harpoons, and 'c' button for tethering/untethering bugs.

DebrisHauler, 2024-05-21:
I did a rebase which pulled in the 2025 sonic themed menu! It also fixed the row of dirt tiles at the top of the screen.
https://github.com/AEFeinstein/Super-2024-Swadge-FW/assets/6497989/b444451f-5d18-49e1-89ff-cf307496f7dc

AEFeinstein, 2024-05-22:
I'm not done working with the visuals of the menu, but it's not gonna change _too_ drastically from here.

DebrisHauler, 2024-05-23:
Did some nice tile work today in aseprite. 
video:
https://github.com/AEFeinstein/Super-2024-Swadge-FW/assets/6497989/c6f20cb7-984c-4d14-aedf-3ff4be17ecf4

Here is also an image of where the medium and hard densities are headed. They should all fit side by side nicely... eventually.

images:
[image]
[image]

DebrisHauler, 2024-05-23:
> I'm not done working with the visuals of the menu, but it's not gonna change _too_ drastically from here.

I appreciate your involvement! It would be cool if the balls orbit around the center of the main menu!

AEFeinstein, 2024-05-23:
Yep, working on animations, better fonts, better colors, and a better battery indicator

DebrisHauler, 2024-05-29:
Progress with some shading and the medium hardness tileset.
video:
https://github.com/AEFeinstein/Super-2024-Swadge-FW/assets/6497989/aedb3731-0f99-4419-8276-5717c041e55f

AEFeinstein, 2024-05-29:
That's looking so trashy!

DebrisHauler, 2024-05-29:
I made a procedural way to generate levels in blender. The pixels are read in by the game to set soft, medium, or hard density tiles.

[image]

While it's great to see each tile set works seamlessly beside the others, they are as visually distinctive as I was hoping. For gameplay purposes it would have been good if the player could very clearly distinguish blocks that will require more damage to delete. hmmmmmmmmm

https://github.com/AEFeinstein/Super-2024-Swadge-FW/assets/6497989/4362add5-4045-4612-afac-4e2e079695e6

AEFeinstein, 2024-05-29:
Color could be a strong indicator, and the tiles seem pretty grey. There's some nice goop already which could be jazzed up.

DebrisHauler, 2024-05-29:
> Color could be a strong indicator, and the tiles seem pretty grey. There's some nice goop already which could be jazzed up.

I might do something more with the goop. Using the same color palette is kind of important for the illusion of objects crossing the tile seams.

DebrisHauler, 2024-05-30:
Did a despeckle filter on all the tiles. Got Ovo bouncing nicely. Digging only occurs at steep angles. Camera is no longer nauseating while you dig.

https://github.com/AEFeinstein/Super-2024-Swadge-FW/assets/6497989/940fed20-077b-4bb1-9862-25609e6e2092

AEFeinstein, 2024-05-30:
For color, as long as the seams are consistently grey, color inside the tile can be whatever. Just a thought. Or you could make a ton of tiles for each transition at each edge. That seems less tenable.

AEFeinstein, 2024-05-30:
Maybe have the tiles change in density as you chip away at them too? Might be nice feedback

DebrisHauler, 2024-05-30:
> For color, as long as the seams are consistently grey, color inside the tile can be whatever. Just a thought. Or you could make a ton of tiles for each transition at each edge. That seems less tenable.

Good idea, I'll try eventually. Thank you! I may move on to other gameplay pillars first next jam session.

DebrisHauler, 2024-06-10:
https://github.com/AEFeinstein/Super-2024-Swadge-FW/assets/6497989/8057f9c3-2eb5-4dcf-a706-9a044db109d5

DebrisHauler, 2024-06-10:
Did lots of work on tile art today shown in the video above. The midground tiles will be regions restraining x,y traversing bugs. If you expose an entire midground region from the foreground, I plan a flood fill style collapse neutralizing any bugs there.

I've also started some animation in blender for when you bust tiles. Check it out!

https://github.com/AEFeinstein/Super-2024-Swadge-FW/assets/6497989/91450582-2e41-453e-ae89-b95c996b4c92

AEFeinstein, 2024-06-10:
Whoa animation. I'm all programming, no art, so this is very impressive to me!

DebrisHauler, 2024-06-15:
Boom, animated entities!

https://github.com/AEFeinstein/Super-2024-Swadge-FW/assets/6497989/e77fdbcb-8e25-4f8c-8e9e-dd43849a93dd

DebrisHauler, 2024-06-15:
I'm trying to think up a method to crumble midground regions when they become unsupported by foreground. An exhaustive floodfill upon every midground reveal sounds expensive. Thinking of some behaviors that are present in valheim. Either managing the search over many frames or storing some support data in every tile. I don't know yet.

DebrisHauler, 2024-06-15:
Terrain crumbling is a fairly important mechanic to me because it would open up different strategies to deal with enemies.

AEFeinstein, 2024-06-15:
Hm, I don't quite get what you mean by midground and foreground. Is the game taking place on one 2D plane or more?

DebrisHauler, 2024-06-15:
There is a another plane of gameplay rendered behind the player. Right now it's looking like a pale gray. So I was thinking of some bugs that can traverse x & y axes on that plane using a top down rendering. And some bugs that can only patrol left and right on the foreground using a side rendering.

AEFeinstein, 2024-06-16:
If I'm understanding, you have two 2D grids that are somehow connected along the Z axis, and you're looking for wholly disconnected sections to crumble? Flood fill is going to be your best bet, and I would strongly recommend implementing it with a `list_t` rather than recursion. Recursion has a lot of function calling overhead (slow) and can chew through the stack (big). With a `list_t` you can:
* Add your initial index to the `list_t`
* While the `list_t` is not empty:
    * Remove an index from the `list_t`
    * Check that index, and mark it as checked & connected (every space needs a boolean flag)
    * Add all that index's unchecked neighbors to the `list_t`

DebrisHauler, 2024-06-17:
Sounds spot on. Will be a while again for me to work on things. My main drive has died today and I'll have to setup everything again from scratch.

AEFeinstein, 2024-06-17:
Oh yikes. I hope you had everything pushed to GitHub

DebrisHauler, 2024-06-18:
Well this project is in a decent place. I only lost some blender and aseprite work for tilesets and level gen since I was only checking in final assets.

DebrisHauler, 2024-08-23:
I'm overdue for a blog post! TLDR: I got my dev environment set up again, and even got to debugging on physical swadge hardware over UART & PuTTy thanks to all the other super helpful advice I'm picking up from Adam, Dylan, CNLohr, and others too. That is not easy. Got some awesome graphics & lighting done on the background, midground, and foreground. And got character rotation implemented.

There is some awesome new background art. Concept art was provided by Kaitie, and I went over it in Krita with a two tone lighting style, and dithered gradient in the lower depths with Python Image Library. Also came up with another background, so there are two layers back there at different parallax distances if you look closely.

This cycle was a big push towards immersive lighting, and I am excited to show you! I will be going forward with a battery stat the player must manage which doubles as health (rather than having a health bar). You will only dive as long as your headlamp is powered on to show you the way and your battery also keeps your machine airborne. Taking damage would also knock down your time remaining in the level by "routing energy into the shield".

I made this lookup texture where the red, green, and blue channels represent how the headlamp illuminates the level as garbotnik rotates from left to right. It represents left facing light, forward facing, and right facing. The left and right facing are also positioned to light up a little bit behind the player. Maybe this could be attributed to bounce light, but mainly I thought it would be frustrating gameplay to have complete darkness directly to your left or right. Technically I might have sampled the red channel backwards for right-facing situations, but this was a bit quicker to code using the three channels. As you rotate towards center, the midground interpolates more or less interpolates towards darkness, and then the foreground interpolates towards the blue channel and offsets that circular lookup left or right based on character rotation.

https://github.com/user-attachments/assets/571a50f1-a1c0-49b2-9b54-86508d4ee8ab

[image]

You may notice the tiles are lighting on a quarter-tile size with a dithering effect. It was a real puzzle to figure out this effect with the limited swadge color space, but I used python image library to chop up my full-size tiles and generate 6 carefully dithered versions of everything for brightness levels zero through five. This produced 1,080 files totaling 203 KB even after optimizing the tile lookups to reuse certain sprites. (We get 4 or 16 MB of space, right? This should be OK). The code loads them into spiram via for loops (which I am told I basically cannot possibly fill via game dev). This takes a good minute on the swadge, or 6 seconds in the emulator. I'll ask for some help adding a loading screen most likely.

[image]

Here's what the midground and background look like!

https://github.com/user-attachments/assets/a32f99f5-f4d3-4d25-b736-425a914844e3

The garbage crumbling algorithm or twin-stick shooting may be next on my list. Crumbling is halfway done already, but it is still needing a* pathfinding to determine stability and some management of uncalculated blocks. Twin stick shooting I imagine will consist of harpoons launching from one of those springy accordion comic hands that extends from behind the player sprite in the direction of a touchpad release event.

Here is a sneak peak of some bugs and pangolin customer animation from Kaitie. I'd like to run them through PIL to generate the different lighting levels if space & load time will allow. I might replace some pixels with bright colors because these would be the fully-lit versions, with 5 darker versions also generated.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

I look forward to hosting a midi music jam in October once the game is in a further state. And development may continue to be in weird concentrated bursts as I also handle some IRL things.

AEFeinstein, 2024-08-23:
I've never been so excited to have this many bugs in the firmware.

AEFeinstein, 2025-01-02:
Gonna keep this open until at least `bb_EnterMode()` is used instead of `bb_EnterModeSkipIntro()`

closed 2025-01-20 by AEFeinstein