AEFeinstein/Super-2024-Swadge-FW #224: First-boot Tutorial / Splash Screen

dylwhich, 2024-05-10:
**Summary**
This would be a mode that automatically starts on the first boot of the swadge (after the factory test is complete) and gives a brief tutorial for the basic functions of the swadge.

**Technical Spec**
This mode would work similarly to the factory test, in that there would be an NVS flag set once the tutorial is complete, and if not set, the tutorial opens instead of the main menu.

The implementation of the tutorial itself could use a generified version of the trigger-based tutorial system used in MFPaint. Throughout the flow of the tutorial, the user would be required to actually perform the actions being explained in order to proceed. For example, in order to complete the tutorial, the user would need to hold down the select button long enough to exit. This would help make it much harder for a new user to immediately get stuck in a mode and have to restart the swadge to exit.

The tutorial should cover all inputs that will be used by modes, and also cover common design patterns used by the modes. This might require a bit more standardization, e.g. having a specific icon that means "you can use the touchpad here" or "turn the swadge like a steering wheel" to be used by the modes.

**UI**
The UI should at least include a main section that shows a rendering of the swadge, or at least its inputs, to visually demonstrate the actions required to complete the next step of the tutorial. Then there would be a section of text for the step's instructions, and maybe a title for each step if it's needed and there's room.

Possibly a stretch goal: use a 3D model for the swadge, so we can have it fly into the screen with a cool animation and sound at the start, and could also zoom in on an area when we want more detail.
 
**Mockups**
There's a bit of prototype work already in the `intro-tutorial` branch with a rough idea of the UI and the tutorial trigger system implemented. TODO: take screenshots and make mockups.

**How to Test**
For judging the effectiveness of the tutorial, we'd probably want some people who aren't familiar with the swadges to try it out uninitiated and give feedback.

AEFeinstein, 2024-05-10:
My wife, the tester!

AEFeinstein, 2024-09-13:
@dylwhich this seems like low-hanging fruit to finish up. By my account this still needs:
* IMU lesson
* Touchpad lesson
* Microphone lesson

And then shim a link to it in "Utilities" for funsies.

AEFeinstein, 2024-11-12:
@dylwhich I branched off your branch and am finishing this off. It's a bit of a hackjob, but it's coming along well.

emilyanthony4244, 2024-12-04:
Hi gang, would it be possible to add a note during the tutorial that the low ringing from the speaker is normal or something at high volumes? I don't want folks to think their Swadge is broken.

AEFeinstein, 2024-12-04:
Sure

AEFeinstein, 2024-12-05:
Added

closed 2024-12-05 by AEFeinstein