AEFeinstein/Super-2024-Swadge-FW #227: Custom Waveform / Sample Playback in Songs

dylwhich, 2024-05-19:
**Summary**
This issue describes enhancements to the existing song player and synthesizer that would allow playing custom waveforms or sample data as part of a song. This would effectively create a way to define a "sound font" that could be used for playing back a track in a MIDI file, without necessarily having a unique sample for each sound or pitch variation.

**Technical Spec**

The implementation would be mostly defined within the synth oscillators, with some helper functions in the song player to configure the source for each channel. I think this is the most straightforward place to put it in terms of ease of use for any mode, and for eventually loading the default track config from a song file itself.

***SW Synth:***

The SW synth would be updated to add two new options for an oscillator. The first would be a custom waveform option, which will work exactly the same as the existing wave functions except with the wave function being provided by the client. The synthesizer would still control the frequency by changing the step size, as with the built-in wave functions.

`void swSynthInitOscillatorCustomWave(synthOscillator_t* osc, waveFunc_t waveFunc, uint32_t freq, uint8_t volume)`

The second option would be a custom sample option, which would be responsible for generating the final sample output on its own. This function would be provided the full accumulator state as well as the frequency value, and a user-defined data (`void*`) pointer. This would allow us to create generic functions capable of transforming existing samples (e.g. pitch-shifting, playing samples with a lower rate) or generating audio data entirely on-the-fly.

`typedef int8_t (*sampleFunc_t)(oscAccum_t accum, uint32_t freq, void* data)`

`void swSynthInitOscillatorCustomSample(synthOscillator_t* osc, sampleFunc_t sampleFunc, uint8_t volume, void* sampleFuncData)`

***Song Player:***

The song player would get new functions for configuring an oscillator's playback type:

`void spkSongSetOscShape(uint8_t songIdx, uint8_t oscIdx, oscillatorShape_t shape, waveFunc_t waveFunc)`
(`waveFunc` would be ignored unless shape were set to `SHAPE_CUSTOM` here)

`void spkSongSetOscSample(uint8_t songIdx, uint8_t oscIdx, sampleFunc_t sampleFunc, void* sampleFuncData)`

DebrisHauler, 2024-05-19:
I would love to dynamically tweak the playback tempo in realtime from my mode. I don't know yet if that's something you would need to design for.

AEFeinstein, 2024-05-19:
I like this, and we need to think of this design from two angles.
1. What does our sound file format look like? Currently it's two series of note & time pairs, one for each buzzer. That's clearly insufficient. At a minimum we need a new, MIDI-like (or MIDI-lite?) format which has event-time-parameters tuples. This would allow us to do things like pitch shift or tempo shift or whatever else. One could also bake the custom waves in there. Sky's the limit. @dylwhich, an instruction set may be a good place to start.
2. How is the sound file generated? Currently it's converting the first two non-empty MIDI tracks, un-overlaping notes, and throwing out everything but the notes. The easy thing to do would be to only copy things that MIDI supports, though I don't know how well that lines up with asks. MIDI files don't store wave data AFAIK, and keep in mind that the solution we come to can't be _too technical_, because it needs to be easy enough to use by non-programmer composers.

Aside from that, I don't love the concept of `swSynthInitOscillatorCustomSample()` because it sounds like it separates where data is stored to play a song. All the data should be in the file. I don't know if it's perfectly in line with your thinking, but I'd rather have a separate callback function that sits between the synth output and the speaker to transform outgoing samples. One could use that to run samples through a low pass filter or an echo filter or something else that would be affected by in-game state.

If you want to generate audio data on the fly, don't use the swsynth at all! You can already take over that callback in the swadge mode struct to do whatever you want.

dylwhich, 2024-05-20:
> I like this, and we need to think of this design from two angles.
>
>     1. What does our sound file format look like? Currently it's two series of note & time pairs, one for each buzzer. That's [...]

That's a good idea, and starting from there would probably help with the other issues. I might have focused a bit too much on backwards compatibility and reusing the synth functionality here. Adding instrument and volume controls to each note at the very least would go a long way towards making the rest of this easier to conceptualize.

> 2. How is the sound file generated? Currently it's converting the first two non-empty MIDI tracks, un-overlaping notes, and throwing out everything but the notes. The easy thing to do would be to [...]

Yeah, I was originally thinking we could just open up more tracks per song, with each track having an assigned waveform/sample. Then the song would just use one track for each instrument, and we could easily swap out instruments by changing the waveform or sample for a particular track. But that's not going to scale to lots of instruments very well, and that's also not really what tracks are for. So, different instruments on the same track are a must.

> Aside from that, I don't love the concept of `swSynthInitOscillatorCustomSample()` because it sounds like it separates where data is stored to play a song. All the data should be in the file. I [...]

I probably should have expanded on this a bit more. My thinking was that (as far as the synth is concerned) the sample function is just another wave function, which happens to return sample-shaped waves. And this is how we'd hook in sound fonts, and there would be a lot of helper functions around that to make it easy for modes to use, possibly including a separate file format for them. So a mode wouldn't usually use this directly, instead there would be a helper (something like `setupSoundFont(const char* name, uint8_t songIdx, uint8_t trackIdx)`) that takes care of loading up a sample and then calling `spkSongSetOscSample(songIdx, trackIdx, getSamplePitchShifted, sampleData)`, where `getSamplePitchShifted()` is another helper function that handles shifting a sample into the pitch that the song wants. Or it could set up a totally software-generated sound font, like a drumkit that plays a different drum for each note, just by setting up the oscillator with a different sample function.

But to your point, the mode would have to do that for each track on its own with that implementation, and what I just described isn't actually a sound font, more like just an instrument definition. And the thing earlier about having tags I do think a MIDI-style separation of the song's notes from the song's instruments is a good idea, perhaps it would also be good to let the song specify by default which instruments it wants, and let the mode swap out the instrumentation

> If you want to generate audio data on the fly, don't use the swsynth at all! You can already take over that callback in the swadge mode struct to do whatever you want.

That's fair! I might have covered it already but just to be clear this wouldn't really be intended as a way to just play audio data on its own, you'd have to use it as part of a song. And I think for that case there's definitely some value in being able to synthesize a sample for a particular note without having to deal with the rest of the song playback and timing, etc. And also I want to have a drumkit, which we definitely won't have room for if each one needs its own samples.

-------

All that said, I'll take another pass at this and try not to focus too much on the existing synth implementation. New goals:

* Design a new MIDI-lite song file format that makes sense for this. Or: figure out if we can just use actual MIDI? Maybe we can strip out everything we don't support and heatshrink it and still end up with something reasonable. It should support two tracks, arbitrary instruments per track, and per-note settings for note, volume, instrument, and pitch bending.
* Create a nice interface for implementing and using sound fonts, and defining an instrument programmatically. This could be a file format as well, but for now I think coming up with structs and functions is good enough, and later we can work out a file format once the details are all ironed out.
* Redesign for the song player as something that explicitly supports instruments / sound fonts. Or should it be a new thing entirely? I'm not sure how much of the song player is just for backwards compatibility with the buzzers.

AEFeinstein, 2024-05-20:
Don't worry too much about backwards compatibility for now. The old inputs are MIDI, so it won't be hard to reprocess them into the new format.

Does MIDI support changing instruments mid track? I'm not sure that's necessary. 

I still like the simplicity of NES style synthesis with oscillators only, but I'm curious to see where this sound font thing goes. Maybe we just have a fixed set of instruments for all songs, in addition to the standard waveforms? Would be simpler.

dylwhich, 2024-05-21:
Yeah, the more I'm looking into this the more I'm leaning towards "make all the instruments with oscillators" and having sampled instruments be a stretch goal, or at least leaving that for later.

Also it would be really cool to be able to have a custom instrument where you can record the sample directly from the mic on the swadge, and I'm definitely going to try that eventually. We almost definitely wouldn't be able to save it to NVS or anything but even so it would be really neat. That is, assuming we can switch between the mic and speaker without swapping modes, which I think we can?

> Does MIDI support changing instruments mid track?
It does! The wire protocol calls this a "program change" event. It looks like you can do this individually on 16 channels too. [This page](https://computermusicresource.com/MIDI.Commands.html) has a pretty good overview of the basic commands without much fluff.

There are also SystemExclusive (SysEx) messages which we could use to implement anything else we want, with the caveat that we'd need to get allocated a unique SysEx ID for [$240 per year](https://midi.org/membership) from the MIDI association. Not too steep but also I can't think of a reason it'd actually be necessary either.

> Maybe we just have a fixed set of instruments for all songs, in addition to the standard waveforms? Would be simpler.

I think that would work! And that seems to be kind of in line with [General MIDI](https://en.wikipedia.org/wiki/General_MIDI) which I've been eyeing as a sort of guideline, even if we only implemented some of it. Plus, it also specifies an instrument for square waves and sawtooth waves under the "synth lead" section, and we could probably stick the rest of the waves in there too.

I should probably just go make a new issue specifically for a MIDI investigation/implementation roadmap actually...

dylwhich, 2024-05-22:
Here's the MIDI issue: #230

AEFeinstein, 2024-05-22:
Well rather than doing full MIDI, my reduced instruction set would be:
* [Set tempo](http://midi.teragonaudio.com/tech/midifile/tempo.htm)
* [Set waveform/instrument for channel](http://midi.teragonaudio.com/tech/midispec/pgm.htm)
* [Set volume for channel](http://midi.teragonaudio.com/tech/midispec/vol.htm)
* [Note On](http://midi.teragonaudio.com/tech/midispec/noteon.htm)
* [Note Off](http://midi.teragonaudio.com/tech/midispec/noteoff.htm)
* [Pitch Bend](http://midi.teragonaudio.com/tech/midispec/wheel.htm)

Like MIDI, 16 channels seems to be the right number, though I wouldn't have polyphony within a single channel.

I don't really want to get into ADSR envelopes or other effects like tremolo. Is there anything else you'd want?

DebrisHauler, 2024-05-22:
I can compose a whole lot with that! I'm surprised there would even be pitch bend! With what is proposed - either via setting the instrument or waveform itself, or by rapidly adjusting volume of a single held note, I would be able to achieve an envelope effect. Basically a note that hits loudly then decays exponentially to emulate a natural plucked string sound. https://en.m.wikipedia.org/wiki/Envelope_(music)

Exposing all your suggested midi features to be overridden by api calls at every frame opens up the floodgates for procedural music and dynamic music that responds to gameplay. Believe me, people would perturb tempo, pitch, volume, and even set notes on and off on the fly from mode code. But I would also consider that "nice to have". I'm interested to tie a quiet jingle's tempo to player velocity as a reimagined reference to Dig Dug's sound design.

AEFeinstein, 2024-05-22:
Hm, I'd rather not blow up filesizes with many volume adjustment commands if it's the same envelope used each time...

I hadn't thought of runtime API calls, but that seems doable. Sound design and programming have been separate things, but if you want to blend them, you may have to do some of the API design too.

AEFeinstein, 2024-11-30:
I'm gonna call this done. We still have #299 for efficiency, but I don't think we're adding more core MIDI functionality. @dylwhich, feel free to reopen if I'm off the mark,