AEFeinstein/Super-2024-Swadge-FW #228: Issue with drawWsgSimpleScaled()

DebrisHauler, 2024-05-20:
**Description**
Issue with drawWsgSimpleScaled()

**Expected behavior**
Rendering a sprite with slightly negative y should render a cropped portion onto the screen similar to rendering a sprite with a slightly negative x. It seems the issue is only in the y direction.

**Reproduction Steps**
Scale up a sprite rendered to a slightly negative y position. The sprite is pushed down to start at y=0 instead of being clipped.

**Screenshots**
See sprites in the big bug video that are going off the left side of the screen clip correctly. But sprites going off the top of the screen are pushed down. Issue is only with scaled sprites.
(https://github.com/AEFeinstein/Super-2024-Swadge-FW/issues/221#issuecomment-2119533918)

**Logs**
n/a

**Affected Version**
 - Platform: Windows
 - git branch: main
 - git commit: b36f6fe
 - git full sha: b36f6fe8b5cd21ad33f164a8498ce63d4fe0fc41

**Additional Notes**
n/a

dylwhich, 2024-05-20:
Interesting. It looks like when this happens it does actually clip the sprite, but it does it at the bottom instead.

closed 2024-05-20 by AEFeinstein

AEFeinstein, 2024-06-06:
There ended up being another issue here which I resolved in 7f160072febdaaff1edd98b21afdf132e6416ec2

commit edf98d2e (2024-05-20): Fix drawWsgSimpleScaled() misbehaving on negative Y assets. Fixes #228
PR #229 (2024-05-20): Fix drawWsgSimpleScaled() misbehaving on negative Y assets. Fixes #228
### Description

This fixes the issue with  `drawWsgSimpleScaled()` described in #228. When the Y-offset is negative, the drawing of the line is skipped but the pointer to the WSG data isn't advanced, so instead the image is drawn at Y=0, and however many lines should have been skipped drawing at the beginning are skipped at the end instead.

### Test Instructions

Test by drawing some scaled WSGs with a negative Y offset and making sure they're drawn correctly.

### Ticket Links

Closes #228 

### Readiness Checklist
- [x] [... cut]
commit 02289745 (2024-05-20): Fix drawWsgSimpleScaled() misbehaving on negative Y assets. Fixes #228 (#229)
commit 2f6fda82 (2024-05-23): Fix drawWsgSimpleScaled() misbehaving on negative Y assets. Fixes #228 (#229)