AEFeinstein/Super-2024-Swadge-FW #230: MIDI Support

dylwhich, 2024-05-22:
**Summary**
This feature is about turning the swadge into a real MIDI synthesizer! This would mean, at a minimum, being able to use the swadge as a USB MIDI synth, and playing back MIDI files loaded into the firmware. The implementation would be targeted at implementing [General MIDI](https://en.wikipedia.org/wiki/General_MIDI) to whatever extent is feasible.

**Implementation Plan**
The initial implementation can be very basic, as a proof of concept. TinyUSB comes with MIDI Streaming support, so the first step is to add the descriptors and pull that into a mode. Then, we ignore everything except the`NOTE ON/OFF` messages, and play those using the simple waves that are already supported.

The next step would be creating a way to define an instrument to be synthesized. This is the area where we'd have the most freedom with the implementation, so we could start out with something very basic here too: pick a waveform and define an envelope. Once the rest of the MIDI player is fully implemented at a basic level, we can just go wild making the instruments fancier until we run out of time or CPU, or we end up with a full-blown programmable synthesizer. Some less-than-basic things to do here after the initial implementation would be adding low-frequency oscillators and filters without supporting totally arbitrary synth pipelines.

As for the instruments themselves, General MIDI specifies 128 specific instruments (apparently this is the culprit behind every keyboard ever having helicopter sounds, phone rings, and whatever a "Goblins" is), so coming up with definitions for all of these might not really be feasible, but we could probably get a decent number of them, especially if we make it easy for anyone to define an instrument voice with a simple config file or something like that. I assume you can probably export these from actual synth software? Also note that we can come up with our own instruments and define them in a separate "program bank" from the ones General MIDI supports, so we can fix the glaring omission of Cat Meow from the spec.

Along with the instruments, we would create the MIDI player which is responsible for actually "playing" the instruments with a bank of software synths, mixing them together, applying more filters, etc. This would be used directly by the USB MIDI mode, or controlled by an equivalent to the current Song Player when playing a MIDI file. The initial implementation of this could just control a single synth to play songs without any filters, mixing, or polyphony.

After that basic functionality is complete, the rest of the work is basically just improving and expanding the player to meet the General MIDI specs, which would mainly mean:
* 24-voice polyphony
* Pitch bending
* Note velocity
* Sustain pedal signal

Again, all this is keeping in mind that implementing General MIDI seems like a decent goal to shoot for absent any technical limitations, but isn't a hard requirement and there's basically no downside to just partially implementing it.

For a stretch goal after synthesized instruments are working, it'd be nice to support at least one sample-based instrument. Extra cool points if we have a way to record with the swadge mic and then use that as an instrument.

Another thing that's technically possible is piping the audio back to the computer as a USB audio device. Which I guess would be cool if you want to compose swadge music on your computer and record it or play it back on your actual speakers. But you could also just plug the swadge into some headphones or the line in too...

**Resources**
* USB MIDI 1.0 Specification: https://www.usb.org/sites/default/files/midi10.pdf
* A super basic page on the basic MIDI commands: https://computermusicresource.com/MIDI.Commands.html
* Arduino MIDI synth project, which implements all the complicated synth stuff and is MIT licensed: https://github.com/DLehenbauer/arduino-midi-sound-module
* Another Arduino MIDI Synth project, specifically for ESP32: https://github.com/esp32beans/ESP32USBMIDI

AEFeinstein, 2024-05-22:
Other resource: http://midi.teragonaudio.com/

dylwhich, 2024-05-23:
A couple more resources:
* http://www.philrees.co.uk/articles/midimode.htm
* http://blitter.com/~russtopia/MIDI/~jglatt/tech/midispec.htm

dylwhich, 2024-06-23:
A random thought: it should be possible for the swadge to appear as a USB mass storage device and let the user upload a MIDI file into a fake filesystem in SPIRAM. If it works well, we could even make it usable by other modes for something like uploading a custom level. Or we could use it to make restore/backup a thing. Obviously a MIDI file wouldn't realistically fit in NVS but I think it would still be cool to support user-supplied songs somehow even if it doesn't persist.

AEFeinstein, 2024-06-23:
That feels just involved enough to be an easter egg

AEFeinstein, 2024-11-30:
I'm gonna call this done. We still have #299 for efficiency, but I don't think we're adding more core MIDI functionality. @dylwhich, feel free to reopen if I'm off the mark,

closed 2024-11-30 by AEFeinstein