AEFeinstein/Super-2024-Swadge-FW #240: Pretty Keyboard

johnnywycliffe, 2024-06-28:
**Summary**
The current keyboard is white text on a black background. It should be prettier.

**Technical Spec**
A simple added background with some boxes to keep the text visible on screen. Colors can be set for the shadowboxes and text. Drop in replacement for the old keyboard to not break compatibility.  

**Usage**
Instead of calling `textEntryStart()` call `textEntryStartPretty()` with the following arguments:
- Font pointer
- Max length of the string
- Character buffer pointer
- Background image pointer
- Shadowbox color
- Text color

Otherwise, use textEntryDraw() and textEntryInput() as before.

**How to test**
In the PrettyKeeb branch:
Run ./swadge_emulator.exe -m Keyboard Test

Edit background, text color, and shadowbox color in main/modes/test/keebTest/keebTest.c

**Example**
[image]

AEFeinstein, 2024-06-29:
Snazzy! This keyboard was originally written for an SSD1306 monochrome display, so I'm glad it's getting a fresh coat of paint!

One thing to note is that the cursor blink is way off. `textEntryDraw()` should take `elapsedUs` as an argument and run an accurate timer like:
```c
bool cursorIsShowing = false;
...

// Flip cursorIsShowing every quarter second (250000 uS)
cursorTimer += elapsedUs;
while(cursorTimer >= 250000)
{
    cursorTimer -= 250000;
    cursorIsShowing = !cursorIsShowing;
}

if(cursorIsShowing)
{
    drawLineFast(...);
}
```

johnnywycliffe, 2024-06-29:
I haven't touched the blinking yet lol. I noticed that the end function for the keyboard literally does nothing despite having a comment "Disarms the cursor blink" so I figure that was broken a while ago.

I can look into it,

AEFeinstein, 2024-06-29:
Yeah... I bet it was used in one mode and worked there, so that was that. I'm all for incidentally making it suck less!

johnnywycliffe, 2024-07-06:
Added a few new features:
- Refactored to get rid of a lot of old references (Covered by "swadge2024.h")
- Emphasis color options
- A way to change the text color, background, emphasis color, and shadowbox color after initial start call
- Added blinking, see note below
- Added a new "return" style enter key
- Reformatted code to match more modern styles

Note:
Blinking might be a bit sporadic, I think it's my test program, but `TextEntryInput()` seems to be blocking.

AEFeinstein, 2024-09-13:
@johnnywycliffe is there more stuff you wanted to do here, or is it closeable?

johnnywycliffe, 2024-09-13:
Close it.

closed 2024-09-13 by AEFeinstein