AEFeinstein/Super-2024-Swadge-FW #244: Game - 2048

johnnywycliffe, 2024-07-01:
**Summary**
The most addictive game of... 2014. Let's call this the anniversary edition.

**Technical Spec**
Needs a rock to run on. It's... not hard. Requires only basic functions of the swadge.

**UI**
Uses A or B to accept some results, uses directional controls to move blocks. Start resets the game.

No assets are requires aside from two fonts. An updated splash screen might need more?

**Screenshots**
Title:
[image]

Gameplay:
[image]

Player reaches 2048:
[image]

Game Over:
[image]

**Features planned for 1.1**
- LEDs that color based on highes score block and pulse during a merge
- Sounds
- Better splash screen
- High score saving
- Fix colors of blocks to be easily discerned (Green blocks are too similar)

AEFeinstein, 2024-07-01:
Woo! For the blocks, have you considered using sprites instead of flat colors? That way you could get textures or something cute idk.

AEFeinstein, 2024-07-01:
Also this would be hilarious to play with tilt controls

johnnywycliffe, 2024-07-01:
Movement controls should be easy enough to roll into 1.1.

I'm worried about identification of blocks if I make them too busy with sprites. It might be a good idea to use circles for the higher level ones just so they're visually distinct and then I can reuse colors, but since I already have a problem with recognition, I don't think making a pattern is a good idea.

AEFeinstein, 2024-07-01:
Up to you, though I'll note that for the color-challenged, like myself, differentiating something like stripes v. polka dots is much easier

johnnywycliffe, 2024-07-01:
Wasn't thinking about that. Let me see what I can do.

AEFeinstein, 2024-07-01:
You can also do cool things like use a different, bigger font, and have some blending around font edges. I wouldn't worry too much about the set of tiles. Art can always be filled in later, and with community help if that's your jam :)

johnnywycliffe, 2024-07-01:
I'm very much not an art person. I'll take all the help I can get in the art department.

johnnywycliffe, 2024-07-03:
Version 1.1:
- New, easier to read sprites for the tiles
- Sounds for button clicks and BGM
- A new splash screen with visual interest and high score displayed
- High score saved between resets
- LEDs on the side now illuminate to help show player which direction the blocks moved, or if they didn't move
- LEDs also provide more visual interest in non-gameplay modes

closed 2024-09-19 by AEFeinstein