AEFeinstein/Super-2024-Swadge-FW #256: Tilt Controls for 2048

johnnywycliffe, 2024-07-10:
**Summary**
There are several requests for tilt controls for the 2048 game. They were left out of the initial pull due to an inability to test on my part. Once I get that fixed, I will be implementing this.

**Technical Spec**
Depending on user input, a set of tilt controls can be enabled. Button entry will not be disabled. Player will be able to tilt the swadge to drop blocks, before returning it to level to reset the tilt. Otherwise, it would just constantly dump blocks which wouldn't be very practical. Players will also be able to re-center the IMU as an anti-frustration feature.

On top of the hardware requirements already required for 2048, the IMU will be required.

**UI**
UI
- Main menu will now prompt player to choose to play with or without tilt controls
- When tilt is activated, a small sprite will be drawn on the side of the screen to indicate this. 
- Pause screen will indicate the ability to turn on or off screen
- Pause screen will indicate that the IMU can be re-centered at any time by pressing B

Controls:
- In start screen, A starts a game without IMU controls
- In start screen, B starts a game with tilt controls, setting the current orientation as centered.
- In game, B re-centers deadzone to avoid issues with moving around
- When paused, A toggles tilt controls on or off.
- When paused, B will now exit to start screen
- When paused, Start will now return to game

**How to Test**
Play a game of 2048, press "B" to enter the game, and tilt the swadge around to shift the blocks. IMU re-centering can be tested by moving it way out of previous orientation before pressing B, then tilting the swadge once more. All other changes should be self-evident as visual indications.

AEFeinstein, 2024-07-10:
From #255 
> My Windows PC will not allow for proper communication between my swadges and serial terminals. I can flash swadges, but I can't get them to report acceleration data either.

I don't know how you're trying to report that data, but it may be easiest to use the Swadges's USB HID controller mode. It's already reporting orientation data there, so the trick would be getting the emulator to recognize that as input rather than the keyboard. And at that rate, you could pipe all the USB HID events into the emulator and use the Swadge as a full controller for the emulator.
[image]

AEFeinstein, 2024-07-10:
https://github.com/signal11/hidapi is intriguing

AEFeinstein, 2024-10-18:
I'll put this off until the sequel, 2049

closed 2024-10-18 by AEFeinstein