AEFeinstein/Super-2024-Swadge-FW #264: Fix CNFS warning

cnlohr, 2024-07-20:
```
/home/cnlohr/git/Super-2024-Swadge-FW/main/asset_loaders/heatshrink_helper.c: In function 'readHeatshrinkFile':
/home/cnlohr/git/Super-2024-Swadge-FW/main/asset_loaders/heatshrink_helper.c:31:20: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   31 |     uint8_t* buf = cnfsGetFile(fname, &sz);
      |                    ^~~~~~~~~~~
[895/1015] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/asset_loaders/fs_font.c.obj
/home/cnlohr/git/Super-2024-Swadge-FW/main/asset_loaders/fs_font.c: In function 'loadFont':
/home/cnlohr/git/Super-2024-Swadge-FW/main/asset_loaders/fs_font.c:37:20: warning: initialization discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
   37 |     uint8_t* buf = cnfsGetFile(name, &sz);
```

To remind @cnlohr he has to fix it.

cnlohr, 2024-07-20:
(Also remove extra Check: in accel code)

cnlohr, 2024-07-20:
(Also add UART output to USB)

AEFeinstein, 2024-07-21:
I think you fixed this?

closed 2024-07-21 by cnlohr