AEFeinstein/Super-2024-Swadge-FW #266: Consider usage of `esp_rom_phy_disable_cca()`

cnlohr, 2024-07-30:
Based on another thread in my Discord, the rom function, esp_rom_phy_disable_cca()  was brought up.  I am putting this here as a note - that it might be a way to use good P2P on the swadges, without having to earmuf them so bad with the IGI killer.

The test should be done where the IGI code in esp-now would be removed, in a challenging RF environment where you also disable CCA with, and do a pingstream to the swadge and see if it continues to behave well.

cnlohr, 2024-07-30:
To be clear, I am proposing I do this test. I would like to do it later this month, maybe next month, but will need to do it in a nasty RF environment.

AEFeinstein, 2024-10-18:
@cnlohr status on this?

cnlohr, 2024-11-06:
I have time, but I am worried this will need testing at event.  I can do it there.  So yes, let's kick to next year, and I will do testing at-event.

AEFeinstein, 2025-02-06:
Uh-oh, @cnlohr did you do any secret at-event testing? Is this ticket worth keeping open?