AEFeinstein/Super-2024-Swadge-FW #267: Ultimate TTT: Empty squares are sometimes impossible to navigate to

dylwhich, 2024-08-03:
**Description**
If the bottom-right square in a subgame is empty, but the rest of the bottom row and right column are filled, the cursor cannot be moved from the top-left section to the bottom-right square.

**Expected behavior**
You should somehow be able to move down and to the right simultaneously. This seems to be possible already for the top left and right corners 

**Reproduction Steps**
_Steps to reproduce the behavior and how often it occurs._
1. Fill up the rightmost  column and the bottom row with markers, except for the bottom-right corner.
2. Try to select the corner

**Screenshots**
See this image for an example of the situation, outlined in white - the square with the white circle cannot be accessed:
[image]

AEFeinstein, 2024-08-04:
I could have sworn I accounted for that. It's an odd path, but if you click down it should move down one row, try that whole row, then wrap, try that row, and finally try the middle row. Transpose it for right.
[image]

AEFeinstein, 2024-08-04:
Maybe `tttHandleGameInput()` is overly complicated and also broken :(

dylwhich, 2024-09-04:
Getting back to this in a timely fashion, I tried moving down in that situation and that just wraps it to the top of that column instead. This issue doesn't happen in all rotations of the shape though so it might depend on whether horizontal or vertical wrapping happens first or something?

AEFeinstein, 2024-09-07:
@dylwhich I'll leave this to you to fix, unless you kick it back to me. Hot potato ticket!

PR #298 (2024-10-02): Ultimate TTT AI
### Description

Adds a single-player CPU opponent with 3 difficulties:

* Easy -- Plays a random move
* Medium -- Plays optimally within sub-games, but has no overall strategy
* Hard -- Analyzes the board and tries to play the best overall move, taking into account the best sub-game move, and weighing that against what the opponent's best move would be in the sub-game they would get to move in. Or, if they would get to pick any sub-game, their best move overall.

Also changes `tttHandleGameInput()` to use a slightly different method, which should prevent any squares from becom [... cut]
closed 2024-09-29 by AEFeinstein