AEFeinstein/Super-2024-Swadge-FW #277: Timer Mode

AEFeinstein, 2024-09-13:
**Summary**
Figure out if we want to repeat this from last year or scrap it

**Technical Spec**
_How will this be implemented?_
_What peripherals will be used, and how?_

**UI**
_What are the display screens & menus (if applicable)?_
_What user inputs are there for each screen?_

**Mockups**
_Paste any mockups of screens or anything else here._

**How to Test**
How can we verify the feature is working?

AEFeinstein, 2024-09-13:
@dylwhich thoughts?

dylwhich, 2024-09-20:
Hmm, I think there's maybe two good options here if we want to keep the timer. One is to make it a little nicer and add the simple stuff like beeping when the timer ends, maybe a lap feature, and that's probably good enough. The other option would be to drop the standalone timer mode and instead have a combined timer/dice roller which could be something like a "tabletop game utility"

AEFeinstein, 2024-09-20:
How often is a timer used in tabletop games? Like a chess clock?

closed 2024-11-29 by AEFeinstein