AEFeinstein/Super-2024-Swadge-FW #290: Palette Swapping for WSGs

johnnywycliffe, 2024-09-21:
**Summary**
A feature currently missing is the ability to create a palette a la the NES/SNES and have the sprite's colors swapped out automatically. This feature is one that would likely be useful to many programmers down the road for a variety of applications as it allows for smaller sized games with more variety

**Technical Spec**
Either part of wsg.h (current configuration) or a new file wsgPalette.h if we want this separated out. It provides the following new functions:
- drawWSGPalette(); Automatically draws a wsg with the pallet given
- wsgPaletteReset(): Sets the palette back to default
- wsgPaletteSet(): Sets an individual color
- wsgPaletteSetGroup(): Sets a group of colors to swap
- wsgPaletteGenerate(): Run to set the colors before using them

**How to Test**
Go to any mode drawing wsgs

```C
// In modeData_t
{
    wsgPalette_t pal;
}

// In modeEnter
{
    // Palette setup
    wsgPaletteReset(&palette);
    wsgPaletteSet(&palette, c000, c555);
    wsgPaletteGenerate(&palette);
}

// Where the WSG is drawn
{
    drawWsgPalette(&wsg, x, y, &palette, vertFilp, HorFlip, rotation);
}```

AEFeinstein, 2024-09-22:
Conceptually this is good. If anyone starts to run into performance issues, it may be worth considering recoloring the source WSG once, rather than swapping palette on each draw. That would increase RAM usage if the same WSG gets recolored twice in the same frame because you'd need it twice in RAM instead. Food for thought.

closed 2024-09-22 by AEFeinstein

johnnywycliffe, 2024-09-22:
I was going to do that, but in the end a simple map adds a single cycle per pixel. Not nothing, but not much.

It really depends on if we end up speed limited before we end up RAM limited, and for now it seems we have overhead in both.