AEFeinstein/Super-2024-Swadge-FW #299: Improve MIDI Playback Performance

dylwhich, 2024-09-27:
**Summary**
The MIDI engine hasn't been optimized yet and we're running low on cycles so we should take a look

**Technical Spec**
List of random things to look into later, vaguely in order of how much I think they'll help with no basis:

 * Move ADSR logic into sw synth, because it kind of already does this a bit anyway and will probably simplify the logic
 * Clean up the logic that decides whether to check for the next event. It's pretty complicated and it might be faster to just always do the check once
 * Check if all those `__builtin_ctz` and `__builtin_popcount` calls are actually turning into single instructions
 * Only sum oscillators for voices that are actually active (but this might allow "heavier" parts of songs to slow stuff down vs. always doing the same amount of work regardless)

AEFeinstein, 2024-09-27:
I'd also recommend using https://github.com/KDAB/hotspot to profile the emulator and see where the cycles are going. It's not the same as the firmware, but it should be close enough to find any big offenders. I used it when optimizing texture lookups for the raycaster last year.

The makefile target for generating code coverage data can also be kinda helpful for profiling a specific function be looking at the number of times each line is run

AEFeinstein, 2024-09-28:
Some idea of spots to look at
[image]

AEFeinstein, 2024-12-27:
Good as we're getting!

closed 2024-12-27 by AEFeinstein