AEFeinstein/Super-2024-Swadge-FW #302: Pulsar Panda

johnnywycliffe, 2024-10-11:
**Summary**
Fly through space in your little fighter ship as Panda McGraff, blowing up the forces of the evil Oinkerny. Shoot down entire formations, keep your buddies alive, and figure out what Oinkerny's forces are hiding.

Star Fox? What's that?

**Technical Spec**
This game would rely on a 3D renderer. It wouldn't implement any sort of complex shadows, textures, or advanced techniques do to the inherent limits of the ESP32-S2. This game is largely dependent on the 3D renderer running fast enough on enough triangles to create a game. The original Star Fox ran abysmally, but this is 2026, not 1993. Presumably, anything over 20-30 FPS would be acceptable.

An entire 3D Render pipeline will be created for this game.

Game will play similarly to Star Fox on the SNES. 

[image]

Game will require:
- Buttons/C-stick/Tilt for movement controls
- A to fire lasers
- B to fire a bomb
- Maneuvers can be mapped to whichever of C-stick and directional buttons aren't being used to steer.

AEFeinstein, 2024-10-11:
You know we had a 3D flight sim a few years? It was just wireframe renders though. You could build on that. Or DIY, up to you.

johnnywycliffe, 2024-10-11:
Yep. Played it a bit.

Still want to tackle it myself if I can.

johnnywycliffe, 2025-01-05:
After discussion, there's like four or five different 3D engines already. I'm going to amalgamate them since lord knows I'm not the most efficient programmer, and then work on Pulsar.

closed 2025-04-26 by johnnywycliffe