AEFeinstein/Super-2024-Swadge-FW #318: Big Bug crashes after game over dialogue

DebrisHauler, 2024-11-22:
**Description**
Big Bug crashes after game over dialogue. I'm having trouble pinpointing an exact error message.

**Expected behavior**
It doesn't crash.

**Reproduction Steps**
Use the recording csv attached at the bottom of this issue, or just wait for garbotnik to run out of fuel then advance through the game over screens and dialogue with 'a'.

**Screenshots**

**Logs**

**Affected Version**
 - Platform: Windows
 - git branch: bigbug
 - git hash: 822493ed1b2d01314ed7123d88cec88ba2dd231b

**Additional Notes**
Be sure to use the particular git hash when I had fuel decrementing.
[rec-1731865481054.csv](https://github.com/user-attachments/files/17856819/rec-1731865481054.csv)

AEFeinstein, 2024-11-22:
I replayed the CSV on Linux and it's crashing earlier, here:
```c
void bb_setCharacterLine(bb_dialogueData_t* dData, uint8_t index, const char* str)
{
    dData->strings[index] = HEAP_CAPS_CALLOC_DBG(strlen(str), sizeof(char), MALLOC_CAP_SPIRAM);
    strcpy(dData->strings[index], str);
}
```
This is an easy fix, you're allocating `strlen(str) * sizeof(char)`, but that's not enough space for the string's null terminator. You have to do `strlen(str) + 1`.

AEFeinstein, 2024-11-22:
Another pre-bug, from @JVeg199X's MIDI tweaking actually, in `midiSumOscillators()`, `states[voiceIdx].on` should be `states->on`. It's not an array, it's just a pointer to a single variable.

AEFeinstein, 2024-11-22:
Getting to the actual crash, it seems to be that you're setting data to a `bb_physicsData_t` in `bb_updateGarbotnikFlying()`:
```c
if (gData->fuel < 0)
{
    bb_physicsData_t* physData  = HEAP_CAPS_CALLOC_DBG(1, sizeof(bb_physicsData_t), MALLOC_CAP_SPIRAM);
    physData->vel               = gData->vel;
    physData->bounceNumerator   = 1; // 25% bounce
    physData->bounceDenominator = 4;
    bb_setData(self, physData);
    self->updateFunction = bb_updateGarbotnikDying;
    return;
}
```
and then accessing it as `bb_garbotnikData_t` in `bb_UpdateLEDs()`:
```c
int32_t fuel = ((bb_garbotnikData_t*)entityManager->playerEntity->data)->fuel;
```
`bb_garbotnikData_t` happens to be larger than `bb_physicsData_t` so the read is out-of-bounds. Having `void*` of unknown type is kinda dangerous, so I'm going to add an enum to track what it is.

AEFeinstein, 2024-11-22:
Alright, no more crashing. There's still a lot of places that `void* data` isn't type checked before accessing. You may want to write something like:
```c
void* bb_getData(bb_entity_t* self, bb_data_type_t dataType);
```
which would the expected type and returns `NULL` if it isn't right. You'd still have to do `NULL` checks before dereferencing and fail elegantly.

AEFeinstein, 2024-11-22:
Also run `make format` before committing!

DebrisHauler, 2024-11-23:
Thank you, very much!

closed 2024-11-23 by DebrisHauler