AEFeinstein/Super-2024-Swadge-FW #320: cGrove Grove background doesn'

johnnywycliffe, 2024-11-22:
**Description**
_A clear and concise description of what the bug is._

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Reproduction Steps**
_Steps to reproduce the behavior and how often it occurs._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Logs**
_If applicable, add logs from when the problem occurred._

**Affected Version**
 - Platform: [Windows, Linux, and/or Firmware]
 - git branch: [e.g. `main`]
 - git hash: [e.g. a2a1a77 ]

**Additional Notes**
_Add any other notes about the problem_

AEFeinstein, 2024-11-22:
Yep, this is a size limit. Using some [nifty functions](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32s2/api-reference/system/heap_debug.html#heap-information) we can see how much space there is, both free and the largest block:
```c
#include <esp_heap_caps.h>
#include <esp_log.h>

...

ESP_LOGI("CG", "free size %zu, largest block %zu\n",
    heap_caps_get_free_size(MALLOC_CAP_SPIRAM),
    heap_caps_get_largest_free_block(MALLOC_CAP_SPIRAM));
```
I put that right before `garden_background.wsg` gets loaded and saw:
```
I (277588) CG: free size 358644, largest block 249856
```
Thats about ~360kb free with the largest block at ~250kb. An 800x600 image is 480kb, so you're blowing through both of those limits.

Some quick math says the largest image to draw would be 566 x 424. Or draw a plain green background and draw grass/stump/tree/water sprites on top.

johnnywycliffe, 2024-11-23:
Made the sprite smaller. Blargh.

closed 2024-11-23 by johnnywycliffe