AEFeinstein/Super-2024-Swadge-FW #323: Midi music crashes on some songs

johnnywycliffe, 2024-11-23:
**Description**
When certain midi files are played, they crash when they get to a specific section of the song.

**Expected behavior**
Music should actually play!

**Reproduction Steps**
Load either `Chowa_Battle.mid` or `Chowa_Menu.mid`, let the song play until the emulator crashes

**Logs**
Chowa_Battle.mid:
```
==273178==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x626000011894 at pc 0x55555578e644 bp 0x7fffe32fc8b0 sp 0x7fffe32fc8a0
READ of size 4 at 0x626000011894 thread T1
    #0 0x55555578e643 in midiSumOscillators main/midi/midiPlayer.c:516
    #1 0x55555579207c in midiPlayerStep main/midi/midiPlayer.c:1313
    #2 0x555555792491 in midiPlayerFillBufferMulti main/midi/midiPlayer.c:1373
    #3 0x555555799a55 in globalMidiPlayerFillBuffer main/midi/midiPlayer.c:2481
    #4 0x55555589ca52 in dacCallback main/swadge2024.c:824
    #5 0x55555571ed57 in dacHandleSoundOutput emulator/src/components/hdw-dac/hdw-dac.c:90
    #6 0x5555557322dd in EmuSoundCb emulator/src/emu_main.c:687
    #7 0x55555572f8a4 in stream_request_cb emulator/src-lib/cnfa/CNFA_pulse.c:103
    #8 0x7ffff7c614bd  (/lib/x86_64-linux-gnu/libpulse.so.0+0x2f4bd)
    #9 0x7ffff728d53f in pa_pdispatch_run (/usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-15.99.so+0x4153f)
    #10 0x7ffff7c46082  (/lib/x86_64-linux-gnu/libpulse.so.0+0x14082)
    #11 0x7ffff7292486  (/usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-15.99.so+0x46486)
    #12 0x7ffff72931d7  (/usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-15.99.so+0x471d7)
    #13 0x7ffff7293599  (/usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-15.99.so+0x47599)
    #14 0x7ffff7297739  (/usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-15.99.so+0x4b739)
    #15 0x7ffff7c59a72 in pa_mainloop_dispatch (/lib/x86_64-linux-gnu/libpulse.so.0+0x27a72)
    #16 0x7ffff7c5a0c9 in pa_mainloop_iterate (/lib/x86_64-linux-gnu/libpulse.so.0+0x280c9)
    #17 0x55555572f69b in CNFAThread emulator/src-lib/cnfa/CNFA_pulse.c:90
    #18 0x7ffff7391ac2 in start_thread nptl/pthread_create.c:442
    #19 0x7ffff742384f  (/lib/x86_64-linux-gnu/libc.so.6+0x12684f)

0x626000011894 is located 4 bytes to the right of 10128-byte region [0x62600000f100,0x626000011890)
allocated by thread T0 here:
    #0 0x5555556c8d17 in __interceptor_calloc (/home/jwycliffe/esp/Super-2024-Swadge-FW/swadge_emulator+0x174d17)
    #1 0x5555557571d3 in heap_caps_calloc emulator/src/idf/esp_heap_caps.c:38
    #2 0x555555799931 in initGlobalMidiPlayer main/midi/midiPlayer.c:2453
    #3 0x55555589c2b6 in initOptionalPeripherals main/swadge2024.c:564
    #4 0x55555589c7b5 in softSwitchToPendingSwadge main/swadge2024.c:717
    #5 0x555555757683 in esp_deep_sleep_start emulator/src/idf/esp_sleep.c:40
    #6 0x55555589c235 in app_main main/swadge2024.c:524
    #7 0x555555731188 in main emulator/src/emu_main.c:257
    #8 0x7ffff7326d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58

Thread T1 created by T0 here:
    #0 0x55555566c965 in pthread_create (/home/jwycliffe/esp/Super-2024-Swadge-FW/swadge_emulator+0x118965)
    #1 0x55555572c9a6 in OGCreateThread emulator/src-lib/rawdraw/os_generic.h:374
    #2 0x555555730ba7 in InitCNFAPulse emulator/src-lib/cnfa/CNFA_pulse.c:321
    #3 0x55555572c6cb in CNFAInit emulator/src-lib/cnfa/CNFA.c:85
    #4 0x55555573113f in main emulator/src/emu_main.c:232
    #5 0x7ffff7326d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58

SUMMARY: AddressSanitizer: heap-buffer-overflow main/midi/midiPlayer.c:516 in midiSumOscillators
Shadow bytes around the buggy address:
  0x0c4c7fffa2c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c4c7fffa2d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c4c7fffa2e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c4c7fffa2f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c4c7fffa300: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0c4c7fffa310: 00 00[fa]fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4c7fffa320: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4c7fffa330: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4c7fffa340: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4c7fffa350: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c4c7fffa360: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
[... 86 lines cut ...]
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==274225==ABORTING
[1] + Done                       "/usr/bin/gdb" --interpreter=mi --tty=${DbgTerm} 0<"/tmp/Microsoft-MIEngine-In-af5fvspu.oic" 1>"/tmp/Microsoft-MIEngine-Out-ropgdmn3.ybg"
```

**Affected Version**
 - Platform: Windows / WSL
 - git branch: cGrove
 - git hash: [2d0d396](https://github.com/AEFeinstein/Super-2024-Swadge-FW/commit/2d0d396a2fdf381100e456b0a2785ec723042ba2)

**Additional Notes**
I'm not good at music, but it appears to be something to do with the oscillator!

AEFeinstein, 2024-11-23:
> Load either `Chowa_Battle.mid` or `Chowa_Menu.mid`, let the song play until the emulator crashes

How did you load these? I played both in the dedicated MIDI player (file load mode) and neither crashed

johnnywycliffe, 2024-11-23:
loadMidiFile(); as per usual. I made sure to run make clean to get the new slots made too.

I'll see if I can recreate it on another branch.