AEFeinstein/Super-2024-Swadge-FW #349: Don't crop rotated sprites on hardware.

DebrisHauler, 2024-12-26:
**Summary**
I've noticed rotated sprite appear fully in the emulator, but they get cropped to the unrotated dimensions on hardware. Not really a big deal, but it is noticeable on oblong rotated sprites such as the bigbug rocket attachment arm, or the accidentally headless ants climbing the walls. If this is complicated to rework or a loss in performance, we can just close this one out.

**Technical Spec**

**UI**

**Mockups**

**How to Test**

AEFeinstein, 2024-12-26:
Even in the emulator I'm seeing clipping in the positive X and Y directions, but not in the negative X and Y directions. Exceedingly strange. I'm pretty sure this worked as expected when it was written...

DebrisHauler, 2024-12-27:
Odd, I don't have any clipping in my emulator.

AEFeinstein, 2024-12-27:
Oh, I'm dumb. I was checking rotation in UTT and because it draws one subgame at a time, the new checkered background was drawing over the rotated sprite. to the left and bottom.
[image]
Check your layering? I think the rotation is OK.

DebrisHauler, 2024-12-27:
Hmm, with the bugs or blue attachment arm, their tips vanish into thin air only on the swadge.

DebrisHauler, 2024-12-27:
https://github.com/user-attachments/assets/7e0bea94-2981-47df-a9b5-1f0bfdd1feb9

AEFeinstein, 2024-12-27:
Hrm, after fixing my draw order, it's now clipping in the negative X and Y directions on hardware. Something's still afoot.

AEFeinstein, 2024-12-27:
OK, not entirely sure where the math was getting screwey, but using `int32_t` consistently fixed UTTT (at least). PR is up.

closed 2024-12-28 by AEFeinstein