AEFeinstein/Super-2024-Swadge-FW #355: global heap overflow related to drums

DebrisHauler, 2024-12-30:
**Description**
Still having this issue after `make clean all` global buffer overflow in defaultDrumkitFunc main/midi/drums.c:218 when doing a car arena fight in big bug. Possibly associated with the new car song that plays there.

**Expected behavior**
Complete a car fight without crashing.

**Reproduction Steps**
Touch the car and fly around it for a few seconds to get the crash.

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Logs**
==188989==ERROR: AddressSanitizer: global-buffer-overflow on address 0x5555559ce884 at pc 0x5555557df26a bp 0x7fffe38fc7f0 sp 0x7fffe38fc7e0
READ of size 4 at 0x5555559ce884 thread T1
    #0 0x5555557df269 in defaultDrumkitFunc main/midi/drums.c:218
    #1 0x5555557ebaab in midiSumPercussion main/midi/midiPlayer.c:575
    #2 0x5555557eec85 in midiPlayerStep main/midi/midiPlayer.c:1319
    #3 0x5555557ef088 in midiPlayerFillBufferMulti main/midi/midiPlayer.c:1378
    #4 0x5555557f6723 in globalMidiPlayerFillBuffer main/midi/midiPlayer.c:2491
    #5 0x555555946ded in dacCallback main/swadge2024.c:846
    #6 0x55555577c234 in dacHandleSoundOutput emulator/src/components/hdw-dac/hdw-dac.c:90
    #7 0x55555578f51e in EmuSoundCb emulator/src/emu_main.c:687
    #8 0x55555578cb1e in stream_request_cb emulator/src-lib/cnfa/CNFA_pulse.c:103
    #9 0x7ffff7c626b4  (/lib/x86_64-linux-gnu/libpulse.so.0+0x2f6b4) (BuildId: c382e5f73e13efe6f45c168f6b20289bd4902128)
    #10 0x7ffff72cf978 in pa_pdispatch_run (/usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-16.1.so+0x42978) (BuildId: 54814b450b49a8cb47b8b93037621c91f233b86f)
    #11 0x7ffff7c471da  (/lib/x86_64-linux-gnu/libpulse.so.0+0x141da) (BuildId: c382e5f73e13efe6f45c168f6b20289bd4902128)
    #12 0x7ffff72d6aae  (/usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-16.1.so+0x49aae) (BuildId: 54814b450b49a8cb47b8b93037621c91f233b86f)
    #13 0x7ffff72d8307  (/usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-16.1.so+0x4b307) (BuildId: 54814b450b49a8cb47b8b93037621c91f233b86f)
    #14 0x7ffff72d86d2  (/usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-16.1.so+0x4b6d2) (BuildId: 54814b450b49a8cb47b8b93037621c91f233b86f)
    #15 0x7ffff72d72e1  (/usr/lib/x86_64-linux-gnu/pulseaudio/libpulsecommon-16.1.so+0x4a2e1) (BuildId: 54814b450b49a8cb47b8b93037621c91f233b86f)
    #16 0x7ffff7c5ab3a in pa_mainloop_dispatch (/lib/x86_64-linux-gnu/libpulse.so.0+0x27b3a) (BuildId: c382e5f73e13efe6f45c168f6b20289bd4902128)
    #17 0x7ffff7c5b1a0 in pa_mainloop_iterate (/lib/x86_64-linux-gnu/libpulse.so.0+0x281a0) (BuildId: c382e5f73e13efe6f45c168f6b20289bd4902128)
    #18 0x55555578c922 in CNFAThread emulator/src-lib/cnfa/CNFA_pulse.c:90
    #19 0x7ffff73d2a93 in start_thread nptl/pthread_create.c:447
    #20 0x7ffff745fc3b in clone3 ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78

0x5555559ce884 is located 8 bytes after global variable 'bakedDrumsLens' defined in 'main/midi/bakedDrums.c:9189:15' (0x5555559ce7c0) of size 188
0x5555559ce884 is located 28 bytes before global variable 'DONUT_0_SAMPLES' defined in 'main/midi/bakedDrums.c:9194:21' (0x5555559ce8a0) of size 53452
SUMMARY: AddressSanitizer: global-buffer-overflow main/midi/drums.c:218 in defaultDrumkitFunc
Shadow bytes around the buggy address:
  0x5555559ce600: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x5555559ce680: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x5555559ce700: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x5555559ce780: 00 00 00 00 f9 f9 f9 f9 00 00 00 00 00 00 00 00
  0x5555559ce800: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04
=>0x5555559ce880:[f9]f9 f9 f9 00 00 00 00 00 00 00 00 00 00 00 00
  0x5555559ce900: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x5555559ce980: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x5555559cea00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x5555559cea80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x5555559ceb00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
Thread T1 created by T0 here:
    #0 0x55555567fdc5 in pthread_create (/home/jalbracht/esp/Super-2024-Swadge-FW/swadge_emulator+0x12bdc5) (BuildId: fafaf624aea5b60d987968b45d6158d37b591992)
    #1 0x555555789c55 in OGCreateThread emulator/src-lib/rawdraw/os_generic.h:374
    #2 0x55555578de42 in InitCNFAPulse emulator/src-lib/cnfa/CNFA_pulse.c:321
    #3 0x55555578997a in CNFAInit emulator/src-lib/cnfa/CNFA.c:85
    #4 0x55555578e3c6 in main emulator/src/emu_main.c:232
    #5 0x7ffff73601c9 in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #6 0x7ffff736028a in __libc_start_main_impl ../csu/libc-start.c:360
    #7 0x55555565cb84 in _start (/home/jalbracht/esp/Super-2024-Swadge-FW/swadge_emulator+0x108b84) (BuildId: fafaf624aea5b60d987968b45d6158d37b591992)

==188989==ABORTING

**Affected Version**
 - Platform: [Windows, Linux, and/or Firmware]
 - git branch: [e.g. `main`]
 - git hash: [e.g. a2a1a77 ]

**Additional Notes**
_Add any other notes about the problem_

AEFeinstein, 2024-12-30:
I'm not sure which file caused the crash, so I wrote this patch to have the factory test mode run through all bigbug midis. It's taking a while to play, but not crashing yet.
```diff
diff --git a/main/modes/test/factoryTest/factoryTest.c b/main/modes/test/factoryTest/factoryTest.c
index 4d4b153d..c80b4539 100644
--- a/main/modes/test/factoryTest/factoryTest.c
+++ b/main/modes/test/factoryTest/factoryTest.c
@@ -15,6 +15,8 @@
 
 #include "factoryTest.h"
 #include "mainMenu.h"
+#include "mode_bigbug.h"
+#include "soundFuncs.h"
 
 //==============================================================================
 // Defines
@@ -141,10 +143,68 @@ swadgeMode_t factoryTestMode = {
     .fnAdvancedUSB            = NULL,
 };
 
+int midiIdx                  = -1;
+const char* const allMidis[] = {
+    "BigBug_Boss.mid",
+    "BigBug - Car 1.mid",
+    "BigBug - Car 2.mid",
+    "BigBug - Car 3.mid",
+    "BigBug - Collection.mid",
+    "BigBug_Dr.Garbotniks Home2.mid",
+    "BigBug_Dr.Garbotniks Home.mid",
+    "BigBug - Egg 1.mid",
+    "BigBug - Egg 2.mid",
+    "BigBugExploration.mid",
+    "Big Bug Hurry up.mid",
+    "BigBug_Space Travel.mid",
+    "brkBgmCrazy.mid",
+    "brkBgmFinale.mid",
+    "brkBgmPixel.mid",
+    "brkBgmSkill.mid",
+    "brkBgmTitle.mid",
+    "brkGameOver.mid",
+    "brkGetReady.mid",
+    "brkHighScore.mid",
+    "brkLvlClear.mid",
+    "Bump.mid",
+    "Dirt_Breaking.mid",
+    "Harpoon.mid",
+    "r_health.mid",
+    "r_item_get.mid",
+    "r_p_ice.mid",
+    "sh_revenge.mid",
+    "sndBounce.mid",
+    "sndBreak2.mid",
+    "sndBreak3.mid",
+    "sndBreak.mid",
+    "sndBrk1up.mid",
+    "sndBrkDie.mid",
+    "sndDropBomb.mid",
+    "sndTally.mid",
+    "sndWaveBall.mid",
+};
+
 //==============================================================================
 // Functions
 //==============================================================================
 
+void nextMidi(void)
+{
+    midiIdx++;
+
+    if (0 != midiIdx)
+    {
+        unloadMidiFile(&test->song);
+    }
+
+    if (midiIdx < ARRAY_SIZE(allMidis))
+    {
+        printf("Testing %s\n", allMidis[midiIdx]);
+        loadMidiFile(allMidis[midiIdx], &test->song, true);
+        soundPlayBgmCb(&test->song, MIDI_BGM, nextMidi);
+    }
+}
+
 /**
  * @brief Enter test mode, allocate memory, initialize code
  */
@@ -175,14 +235,16 @@ void testEnterMode(void)
     setMicGainSetting(MAX_MIC_GAIN);
 
     // Play a song
-    loadMidiFile("hd_credits.mid", &test->song, true);
+    // loadMidiFile("hd_credits.mid", &test->song, true);
     switchToSpeaker();
-    midiPlayer_t* player = globalMidiPlayerGet(MIDI_BGM);
-    midiGmOn(player);
-    midiSetFile(player, &test->song);
-    player->loop = true;
-    midiPause(player, false);
-    test->spkActive = true;
+    // midiPlayer_t* player = globalMidiPlayerGet(MIDI_BGM);
+    // midiGmOn(player);
+    // midiSetFile(player, &test->song);
+    // player->loop = true;
+    // midiPause(player, false);
+    // test->spkActive = true;
+    bb_setupMidi();
+    nextMidi();
 
     // Clear out accel setting.
     accelSetRegistersAndReset();

```

AEFeinstein, 2024-12-30:
`sh_revenge.mid` is a culprit

AEFeinstein, 2024-12-30:
OK, looks like it's trying to use the `BELLTREE` drum, and that isn't implemented. That had previously been caught by a default case in a switch statement. Nice find.

closed 2024-12-30 by AEFeinstein

AEFeinstein, 2024-12-30:
No, thank you!