AEFeinstein/Super-2024-Swadge-FW #384: Chowa Grove crash

DebrisHauler, 2025-01-07:
**Description**
In Chowa Grove, I selected "reset all game data" and then "play with chowa" crashed.

**Expected behavior**
don't crash

**Reproduction Steps**
select "reset all game data" and then "play with chowa"

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Logs**
_If applicable, add logs from when the problem occurred._

**Affected Version**
 - Platform: [Windows, Linux, and/or Firmware]
 - git branch: [e.g. `main`]
 - git hash: [e.g. a2a1a77 ]

**Additional Notes**
_Add any other notes about the problem_

DebrisHauler, 2025-01-07:
Never mind the description. "Play with Chowa" alone is just crashing.

johnnywycliffe, 2025-01-07:
When you tried to format the game, did you get the prompt to enter your new username?

johnnywycliffe, 2025-01-07:
If not, then it's almost certainly the format function.

AEFeinstein, 2025-01-07:
Neither the swadge nor the emulator are crashing when I follow those steps (23f5667aad21de8a54483bc694ebd46a02f8b44d). Both display the keyboard right on top of the reset data warning (a little visually odd but w/e).

@DebrisHauler perhaps you have something old and stale in NVM that's causing a problem? Either delete `nvs.json` or konami code the main menu and factory reset your Swadge and try again?

johnnywycliffe, 2025-01-07:
Emily had the same issue before as well- I need to track it down regardless.

My intuition says it's because I save the Chowa wholesale to NVS so when it's loaded in a different state (the struct changed) it crashes since data is loaded wrong.  Still, erasing the NVS should handle it.

If I'm right, It shouldn't affect swadges that only have one version of the code on them.

I haven't had a chance to look at the code yet, still trying to relax after work for a few minutes.

johnnywycliffe, 2025-01-07:
...Which is what stale NVS means. I'm tired.

DebrisHauler, 2025-01-07:
Not sure what is different about my environment. I never had the keyboard appearing in this test. I did the konami code, got a cool song. No luck in Chowa Grove. Also deleted nvs.json followed by idf.py build and flash. Still crashing on clicking "Play with Chowa" after that.

johnnywycliffe, 2025-01-07:
Currently unable to reproduce on:
- Emulator
- Gunship
- HDProtoV1

As a note, nvs.json is only for the emulator. Has no bearing on the Hardware.

The factory reset didn't get it? Crazy.

DebrisHauler, 2025-01-07:
I guess we're good then if that's how production devices work. I probably screwed something up.

johnnywycliffe, 2025-01-07:
Just to check, when you are in-mode and reset just the garden, do you confirm with Pause/Start or with "A"?

Because only start works so it's harder to accidentally mash

DebrisHauler, 2025-01-07:
I am not getting to the garden. Start does nothing on "Play with Chowa" and A crashes it.

johnnywycliffe, 2025-01-07:
I'm saying on reset. If you try to delete the chowa data, are you deleting it using start or A?

DebrisHauler, 2025-01-07:
I used A.

johnnywycliffe, 2025-01-07:
Well, that's not going to work

johnnywycliffe, 2025-01-07:
Try again and press start on the delete screen.

DebrisHauler, 2025-01-07:
Okay, that got me into the game. That must be my first-ever mistake, not reading it correctly.

johnnywycliffe, 2025-01-07:
Closing issue, ID 10T error type PEBKAC

closed 2025-01-07 by johnnywycliffe