AEFeinstein/Super-2024-Swadge-FW #391: Mag: The Festing

g-r-a-e-m-e, 2025-01-11:
# Mag: The Festing
Hello! This is my proposal for a Swadge game, ***Mag: The Festing***. The **Summary** section outlines the theoretical capabilities of the game, but should not be considered promised features until development is complete. Thank you for taking the time to look at this proposal :)

***Constructive feedback is welcome!***

___ 

## Summary
***Mag: the Festing***, shorthand MTF, is trading card game with cards based on Magfest themes and memes. Swadge owners can have access to pre-constructed starter decks out-of-the-box, and can battle one another via the Swadge's network capabilities. Cards of different rarities can be collected by entering codes found throughout Magfest (Marketplace, Indie Game Arcade, Panels, etc.). Users can build decks using cards in their collection, and their collections can be further expanded using a player-to-player trade system.

## Technical Specifications
_How will this be implemented?_
_What peripherals will be used, and how?_
| Component | Purpose |
|---|---|
| [Swadge API](https://adam.feinste.in/Super-2024-Swadge-FW/#apis) | Application development |
| 202X Swadge Hardware | `TODO` *Dependent upon hardware revisions* |
| Game Mechanics | How the game is played |
| Card Database | Database of cards incorporated into the game |
| Acquisition System | Methods for users to collect cards |
| Inventory Management System | Methods for users to manage their card collection (search, filter, group, etc.) |
| Trade System | Methods for users to trade cards in their collection |
| Rarity System | Extension of the Acquisition System for "spawning" cards of different rarities |
| Battle System | Player vs. Player battle mechanics |
| ... | ... |

## User Interface
_What are the display screens & menus (if applicable)?_
_What user inputs are there for each screen?_
| Screen | Inputs |
|---|---|
| Main Menu | Navigation (Up/Down buttons, A/Select) |
| Connectivity Menu | Join/Host selection (Left/Right buttons, A/Select), Return (B) |
| Collection Menu | Navigation (Directional pad), Card selection (A/Select), Return (B), Add Card (Directional pad, A/Select, B) |
| Deckbuilding Menu | Navigation (Directional pad), Card selection (A/Select), Return (B), Card Search (Directional pad, A/Select, B) |
| Battle Screen | `TODO` |
| Battle Summary Screen | `TODO` |
| Trade Screen | `TODO` |
| ... | ... |

## Mockups
_Paste any mockups of screens or anything else here._
`TODO`

## How to Test
How can we verify the feature is working?
`TODO`

___

## Revision History
| Version | Contributor | Date | Notes |
|---|---|---|---|
| 0.1.0 | [g-r-a-e-m-e](https://github.com/g-r-a-e-m-e) | 2025-01-11 | Initial feature request |

AEFeinstein, 2025-01-12:
As an ex-avid-MtG player, I can get behind this. The team is in final prep mode, so this won't get much attention until after Mag.

Right now I wouldn't worry too much about the technical aspects. Designing a fun and balanced card game is hard enough, and I'd start there. MtG is good inspiration, but far too complicated. Think about what you can strip away to make it simpler but not boring. Maybe check out the Pokemon TCG game for GBC which crams a TCG into very few pixels.

Also it's worth thinking about our [design principles](https://docs.google.com/document/d/1TzzatyRWp9t26YWF3qUlOg7NFgmsueL-0suqyDT98IE/edit?tab=t.0#heading=h.pa8ndddehvta).

AEFeinstein, 2025-01-12:
Spitballing, it might be neat to make matches a fixed four rounds, for the four days of Magfest (and maybe a fifth for setup Wednesday). It's nice thematically and caps games at not-too-long. That also fits nicely with a "play this card on or after day X" mechanic like Serra Avenger. I personally like time as a resource

AEFeinstein, 2025-01-12:
Instead of five colors, you could do three categories, 'music,' 'gaming,' and 'and.' Could be a rock-paper-scissors thing.

g-r-a-e-m-e, 2025-01-13:
Spitballing is great.

I like the idea of limiting the number of rounds. Maybe each round nets a number of points and a win condition is "most points after *n* rounds," akin to Gwent. 

We have a similar idea for the resources. I'm wondering if the resources available are fixed per round (each "day" of the fest). Maybe something like *x* music resources and *2x+1* gaming resources where players can play cards with a sort of exhaustion mechanic - e.g. *Card 1* requires 2 gaming resources to play and *Card 2* requires 1 music resource, and resources are replenished after a "short rest" or something. 

Pushing the exhaustion idea a bit... maybe players can expend resources from their next turn by "staying up all night" to play cards that require resources greater than their daily replenishment. For example, a card like "Smash Tournament" that requires 6 gaming resources (assuming a max of 5 are available that day), but players' "rest" steps only refill their daily maximum.

Something like a fixed resource pool might prevent players from becoming "mana screwed/flooded," so to speak, as well as simplify the game mechanics, but it might also be easy to exploit.

AEFeinstein, 2025-01-14:
Sleep is a funny resource. I never get enough of it!

AEFeinstein, 2025-02-06:
@g-r-a-e-m-e, happy post-Magfest! I don't know if you've thought more about this, but I haven't much. At this phase we should be focusing way more on game design than implementation. It may be quicker to iterate that with paper.

Anyway, if you're still interested, send an email to circuitboards@magfest.org with a good email address so I can add you to our development Slack.

g-r-a-e-m-e, 2025-02-06:
@AEFeinstein email sent!

AEFeinstein, 2025-03-28:
@g-r-a-e-m-e any more thought about game mechanics?

g-r-a-e-m-e, 2025-03-28:
@AEFeinstein indeed! I'll have some free time this afternoon to follow up.

g-r-a-e-m-e, 2025-03-29:
@AEFeinstein Without getting too long-winded, I have been working through ideas and documenting that process in Obsidian. The core game mechanics have been roughed out and I have been designing the cards and their abilities. I have a simple `.ods` file that I'm using to keep track of the cards, resource cost, point values, and other attributes, and have some high-level statistics put together on the card pool. Once I have a reasonable (>= 50) number of distinct cards, I plan on putting together an index card version that I will be playtesting with a group I routinely play Magic with. I have blocked out some time this weekend to continue work on the cards, but would be happy to provide a more "formal" update sometime next week!

AEFeinstein, 2025-03-30:
Nice! I like that process, especially the index cards.

closed 2025-10-15 by AEFeinstein