AEFeinstein/Super-2024-Swadge-FW #393: Bigbug Fuzzing

AEFeinstein, 2025-01-13:
**Description**
I ran the fuzzer all day and found three crashes. They seem quite rare, so I don't know if you care.

1. `bb_updateBugShooting at /home/adam/git/Super-2024-Swadge-FW/main/modes/games/bigbug/entity_bigbug.c:1499`
	```c
	// call it paused and update frames in it's own update function because this one uses another spriteIdx.
	bb_entity_t* spit = bb_createEntity(&self->gameData->entityManager, LOOPING_ANIMATION, true, BB_SPIT, 10,
		                            self->pos.x >> DECIMAL_BITS, self->pos.y >> DECIMAL_BITS, true, false);
	if (self->gameData->entityManager.playerEntity != NULL)
	{
	    bb_spitData_t* sData = (bb_spitData_t*)spit->data; // CRASH HERE
	```
1. `bb_GameLoop at /home/adam/git/Super-2024-Swadge-FW/main/modes/games/bigbug/mode_bigbug.c:1647`
	```c
	bb_entity_t* radarPing
	    = bb_createEntity(&bigbug->gameData.entityManager, NO_ANIMATION, true, BB_RADAR_PING, 1,
		                bigbug->gameData.entityManager.playerEntity->pos.x >> DECIMAL_BITS,
		                bigbug->gameData.entityManager.playerEntity->pos.y >> DECIMAL_BITS, true, false);
	bigbug->gameData.screen    = BIGBUG_GAME_PINGING;
	bb_radarPingData_t* rpData = (bb_radarPingData_t*)radarPing->data; // CRASH HERE
	```
1. `bb_startGarbotnikLandingTalk at /home/adam/git/Super-2024-Swadge-FW/main/modes/games/bigbug/entity_bigbug.c:4755`
	```c
	void bb_startGarbotnikLandingTalk(bb_entity_t* self)
	{
	    bb_garbotnikData_t* gData = (bb_garbotnikData_t*)self->data; // CRASH HERE
	```

My guess is that for all of them `bb_createEntity()` is returning `NULL` here, and that's not getting checked before dereference:
```c
if (!forceToFront && entityManager->activeEntities == MAX_ENTITIES)
{
    // ESP_LOGD(BB_TAG,"Failed entity creation. MAX_ENTITIES exceeded.\n");
    return NULL;
}
```

**Affected Version**
 - Platform: Linux emulator
 - git branch: `bigbug`
 - git hash: 3945f1830874feecc6d5099d4fca85b89161c77b

DebrisHauler, 2025-01-16:
Thanks, at least I have a clear fix for the first one in mind.

AEFeinstein, 2025-01-16:
It's easy enough to check for `NULL` these three places, but there are many other times entities are created so idk

DebrisHauler, 2025-01-19:
Thanks for the testing.

My commits should have it resolved, hopefully.

16be83be101ccc7e3d7e7e12abf64797efb17552

301c1eb3936ea15265a76c6bdb6388ddc9070b37

4c66cc85d9a158cbf8273fae81a07e513795dea6

closed 2025-01-19 by DebrisHauler