AEFeinstein/Super-2024-Swadge-FW #397: Utility - Trophy System

johnnywycliffe, 2025-01-16:
**Summary**
This utility will provide an entire trophy system for all devs to integrate into their swadges. It is light on processing power, and in the simplest case requires only four functions to operate. It uses very little RAM, avoids hammering the NVS, and has an entire points system a la XBOX for players to be able to compete with.

There are three types of trophies:
- Trigger: Only required to trigger once. These are best suited to trophies that are secrets, milestones, etc.
- Additive: Each time update is called, adds to the tracker until the max value is reached. Useful for lifetime stats.
- Progress: Each time the trophy is updated, it takes the highest value and uses that. This is best for trophies that are expected to be completed in a single run, such as farthest distance climbed in a race or highest amount of coins collected per life.

A banner automatically shows whenever the developer wants it to, can be set to trigger off milestones, and only when it is actually updated to avoid spamming text boxes. The banner's behavior can be set to avoid harming the player's experience by blocking text boxes or showing up for too long.

Additionally, a trophy display list function will be provided so users don't have to make their own, similarly to the menu system used in prior years. A full trophy showcase mode shall also be developed

**Technical Spec**
A simple `#include "trophy.h" loads all the necessary data. The settings for the mode (Speed, duration, whether it scrolls or comes in from the top or bottom) are set in the mode entry. Similarly, each trophy can be initialized in the mode entry, and on subsequent reloads it just exits. 

Developers then update the trophy's status whenever appropriate; the update function handles adding the trophy to the draw queue if appropriate, loading all the relevant assets, and saving to the NVS when appropriate. For instance, if the trophy is already won, there's no reason to display the trophy again.

The only real concession is that the developer is required to draw the trophy draw function at the end of their draw code and that may not be simple if the mode is written in a non-standard way.

A standard font would need to be loaded at all times when the trophy system is loaded to avoid crashes.

Currently utilizing NVS namespaces to de-conflict two or modes utilizing the same trophy name. Two individual modes may both use the title "You beat the game!" without causing errors. However, inside a single mode the trophies must all have unique titles.

As of this writing (15JAN2025) this uses the NVS to store the trophy data once initialized. Data saved is as follows:
- Title of trophy
- Description of trophy
- Image title string (Used because pointers change and storing an entire WSG is insane)
- Type
- Point value (XBOX gamer point score)
- Current status
- Trigger value

Unfortunately, this is pretty big (157 bytes), given the first three are storing strings. Some compression is required for final implementation. Current ides:
- Indexing: Save all the strings as Consts so they get stored in ROM and store indexes to NVS. This would reduce the individual size down to 19 bytes per, but would require the players to specifically make all their const strings accessible outside the mode for the trophy system to use.
- SPIFFs/File storage: Using another part of the Flash memory would be preferable, due to the ESP32S2 having 4MB of the stuff rather than the 16kb provided to the NVS. This would be better, since speed isn't a concern and we would have room to breathe.
- XML: Using a centralized file that the trophy system draws from for all data when the ID is requested. The rest of the data can be stored in NVS as it's the strings that require the majority of the space.

**UI**
There will be a trophy showcase mode that will display all the player facing trophy data, including status (13/100), image either in full color or grayed-out similar to how Steam does it, and the text. Standard menu navigation is required (UDLR, select, back).

When in game, a banner will appear when appropriate (controlled by the devs) that will display the status. This takes no inputs, as it is a popup.

**Mockups**
TODO: Banner example
TODO: Trophy showcase example

**How to Test**
A custom Trophy test mode will be implemented to show it is working by manipulating the data via a menu. There will also be a showcase mode to view trophies that will be provided to the players so they can view their accomplishments.

AEFeinstein, 2025-01-17:
OK, I like this. A lot. Let's figure out a sane, efficient, easy-to-use architecture.

You definitely don't want to be storing a bunch of string data in NVS. The strings are already compiled in ROM, so you'd be duplicating them to NVS, which isn't great. SPIFFS is no longer used for Swadges, so that's not an option.

Off the cuff, I'm leaning towards adding a function pointer to `swadgeMode_t` like:
```c
void (*fnRegisterTrophies)(void);
```
Each mode would be responsible for calling some function like `void registerTrophy(const* trophy_t trophy)` repeatedly in there, where `trophy_t` is all the immutable data like:
```c
typedef struct {
    const char* name;
    const char* description;
    const char* imageName;
    const char* nvsKey; // A key to an NVS blob with the trophy's current status / score / whatever
    uint8_t type;
    // ...
} trophy_t;
```
So when a mode starts up, it gets its trophies registered (i.e. saved in a linked list or something), and when the display case starts up, it can call all the mode's trophy registration functions.

When a thing happens, we'd call something like `void updateTrophy(const* trophy_t trophy, int32_t val)` and that would use the NVS key to update state, display a banner, etc. `val` can be treated appropriately if the type is trigger, additive, or progress.

Also we do a little bit of overdraw on the screen currently. You could draw here for trophy popups too. See:
```c
// Draw 'progress' bar for exiting. This is done right before the TFT is drawn
int16_t numPx = (tHeldUs * TFT_WIDTH) / EXIT_TIME_US;
fillDisplayArea(0, TFT_HEIGHT - 10, numPx, TFT_HEIGHT, c333);
```

johnnywycliffe, 2025-01-17:
The only issue I have with this is portability. I want there to be a mode that collates all the trophies from around the Swadge, and the only way I can see that happening is if this mode is able to include all the files with trophies in them. 

This is leaving location of the data and inclusion into the system up to the individual dev. Maybe if we "force" each dev to put it into a secondary file? Like, a chowa_trophies.h that needs to be included.

I'd just rather it pulled all the relevant data.

dylwhich, 2025-01-18:
There's no reason there couldn't be a mode to list all the trophies, even if they were registered with a function in the mode struct. As long as wherever they get registered to has some way to return all the registered trophies, then any other modes just use that and don't have to worry about what specific file or mode they came from, just include `trophy.h` or whatever.

I think something like what Adam suggested could do this, but instead of the trophy case needing to call each mode's registration function individually, I'd say on startup have the firmware automatically call every mode's `registerTrophies()` function if it has one. That would mean adding a list of all modes to `swadge_2024.c` but that's no problem. We'd have to make sure that's not a performance issue but I don't think it should be since it won't be doing anything static.

So, `trophy.h` would have a few functions:
* `registerTrophy(...)` - registers the trophy and adds it to a list (internal to `trophy.c`)
* `updateTrophy(...)` - updates trophy progress
* `getAllTrophies()` - returns the list of all registered trophies, which the trophy case could use
* `getModeTrophies(modeName)` - returns only a single mode's trophies
* `getTrophy(name)` - finds a specific trophy from the list

So then as far as the mode that wants to declare trophies is concerned, it can really put them wherever it wants. There could be a big const array of  `trophy_t` structs that each get passed to `registerTrophy()`, or there could be individual structs defined, or they could be procedurally generated which would be neat.

dylwhich, 2025-01-18:
Another option would be to use JSON files. I don't know if that would actually be more or less convenient to use than just having them in the code but it would be doable, and still stored in a compact way underneath.

There is a JSON parser in the asset preprocessor already, so we could use that to gather up all files named something like `.trophy.json`, and have it convert the JSON data into a CNFS file that's directly readable as an array of `trophy_t`. So, you could actually just do something like:
```
    size_t size = 0;
    const trophy_t* allTrophies = (const trophy_t*)cnfsGetFile("trophies.trophy", &size);
    size_t trophyCount = size / sizeof(trophy_t);
    for (size_t i = 0; i < trophyCount; i++)
    {
        const trophy_t* trophy = &allTrophies[i];
        printf("Trophy %zu is %s\n", i, trophy->name);
    }
```

Though all of that could just be abstracted away into the same `getTrophy(name)` or `getAllTrophies()` interface.

One way this would be better than the method of registering in the mode struct is that it doesn't use any extra memory at runtime since it's reading the trophies directly out of flash. And it also wouldn't need any extra code to run at startup for registering all the modes. The only caveat is we wouldn't be able to compress the data, but since it's in flash that really doesn't matter - we haven't come anywhere near running out, so better to save on runtime allocations which we already have hit the limits of.

AEFeinstein, 2025-01-18:
> I think something like what Adam suggested could do this, but instead of the trophy case needing to call each mode's registration function individually, I'd say on startup have the firmware automatically call every mode's registerTrophies() function if it has one.

I was thinking the trophy case mode would be similar to the Main Menu mode, which references all the mode structs. This way the trophy case can call all the mode's trophy registration functions, but for all other modes, it'd only load its own trophies. Minimize unused overhead! We'd just have to be careful that the trophy registration function in the mode struct is safe to call without the mode being initialized.

I'm also not against keeping a master list of modes somewhere. The emulator kinda already does this, but we can make it more universal. I dunno what modes besides Main Menu and Trophy Case would care?

> Another option would be to use JSON files. I don't know if that would actually be more or less convenient to use than just having them in the code but it would be doable, and still stored in a compact way underneath.

I'm meh on JSON. The last I used it was for Super Mag Bros fighter data, but that was largely so we could modify and reload JSON for the emulator without recompiling. This was pre-CNFS, so that wouldn't work now anyway. I'd be more in favor of a big const C struct with the trophy data. It'd be just about as readable as JSON and much easier to access without parsing.

johnnywycliffe, 2025-01-19:
I think I agree with Adam's way of implementing the function into the `swadgemode_t` struct. Seems logical with how I want it to not require editing too many other files, just add it to the list of modes. Currently, the menu and the external modes list for the emulator have the large include block of all the modes, so if we could generate a .h that covered all the necessary parts to consolidate that to one file that'd be nice. Make adding new modes a lot easier in the future as well.

As far as saving data, this would cut it down from a full blob to a simple 32 bit number, which is obviously much easier to manage. 

I don't agree with making the drawing function part of the last draw step like the exit mode line, at least not without being able to change a few things. These banners are not exactly small, and I'd like to make it so the user can specify top or bottom of the screen per mode, or choose to block the animation during certain times to avoid causing issues with what the player can see on the screen. Maybe if the trophy registration function also allows for setting a few variables?

Last but not least, it's no inconceivable that the main menu would have a few trophies, such as finding the secret menu or trying all modes. Need to make sure however we register the modes we would be able to instantiate these.

johnnywycliffe, 2025-01-19:
Also had a though on the types of Trophies. Adding a fourth type: Task List. As each task is completed, it sets a bit in saved data. When the saved pattern matches the expected value, it triggers. This would be useful for things like "Play all swadge modes at least once" where the order is indeterminate and we can't use a count because then a user could just open one mode 20 times. Works with up to 32 tasks since we're only saving a 32 bit number.

AEFeinstein, 2025-01-20:
> I don't agree with making the drawing function part of the last draw step like the exit mode line, at least not without being able to change a few things.

That's fine. I was mostly bringing up a comparison to what we already do. The Quick Settings menu also overdraws with impunity. Maybe we can have the trophy registration function in the swadge mode struct also specify drawing options? I don't want to make each mode too responsible though, since that ends up being more work for everyone and less consistent visually. It's a balance.

> Last but not least, it's no inconceivable that the main menu would have a few trophies, such as finding the secret menu or trying all modes. Need to make sure however we register the modes we would be able to instantiate these.

The Main Menu _is_ a mode, just like all the others. It'd have its own trophy registration function.

> Also had a though on the types of Trophies. Adding a fourth type: Task List.

Me likey.

AEFeinstein, 2025-02-06:
This seems to duplicate #100 just a bit, so I'm going to close the older ticket. @dylwhich, feel free to bring any ideas from there to here.

closed 2025-05-22 by johnnywycliffe