AEFeinstein/Super-2024-Swadge-FW #400: Asset Preprocessor Improvement

dylwhich, 2025-01-16:
**Summary**
Improvements to the asset preprocessor to make it easier to create new asset types, reduce boilerplate, and generally make it more usable.

**Technical Spec**

Instead of each asset processor doing the same logic to decide on a new filename, there will be an array of config structs, sort of like what's used for the "raw" file types [here](https://github.com/AEFeinstein/Super-2024-Swadge-FW/blob/main/tools/assets_preprocessor/src/assets_preprocessor.c#L26) that just get compressed. The struct would have:
* `const char* inExt`: The extension of the file to be transformed, e.g. `png`
* `const char* outExt`: The extension of the file that will be created, e.g. `.wsg`
* `enum processType_t type`: The type of processing the file gets. Right now would just be `FUNCTION`, but maybe we'd have another for `EXEC` (to use for external tools)
* `bool (*processFn)(const char* inFile, const char* outFile)`: If type is `FUNCTION`, this would be a pointer to the function called to process the file. The in and out file paths are already set up, so no filename parsing is needed inside the handler.
* (if `EXEC` is implemented) `const char* execCmd`: The command to run for processing this file. If it exits abnormally the file is considered failed. The in and out file arguments would be populated in string arguments, so `"cp %1$s %2$s"` would be a valid one. This might be a good uniform way to handle things like the sokoban python script.

Also, it should detect conflicts. If multiple input files are mapping to the same file, that should throw a fatal error.

AEFeinstein, 2025-01-16:
Thank you for making my code suck less

dylwhich, 2025-01-18:
It's all anyone can do really, just make the code suck a little bit less every day!

dylwhich, 2025-01-18:
Another type of processor we could have, which the trophy discussion just made me think of, is one that handles all the assets of a particular type at once. In the trophy example, this would mean taking a bunch of JSON files that represent a struct, and concatenating (and sorting) them into a single array that becomes a single file in CNFS. I think this could also be useful for something like, e.g. swadge hero, where you could have a second asset processor for `.chart` files that outputs an index of filenames that can be used instead of hardcoding.

AEFeinstein, 2025-01-18:
Hm, I kinda see where you're going, but one-file-in-one-file-out is a clean paradigm. I'd wait until we have a need for concatenation before building it as a generic feature.

dylwhich, 2025-02-19:
Just thought of something else that might be useful, though the use case is maybe a bit convoluted. It would be an improvement to CNFS, but it would also go along with asset processor changes. The idea is to make it easier to reference other assets by name, and convert that name into the CNFS file offset instead of storing the string. And then there would be a CNFS function that gets files by that offset directly, without doing the binary search. And, I guess, a function to get the proper name as well.

I think the utility of something like this came up a couple times in the trophy discussion, where storing the trophy icon name isn't the best use of space.

In the asset processor side, it would be implemented as a function of the new asset management code, which would have to gather up all the files and determine what their output names would be first. Then, it could actually call the asset-specific processors. The processors would be able to call a function like `int getAssetCnfsIndex(const char* assetName)` and use the result wherever the filename would have been used.

There is the one drawback of this, which is just that you would _have_ to use the asset preprocessor to use these references, since there's no way to get the CNFS file offset when defining a static struct. I guess you could also do some union shenanigans to use either an offset or a filename but that seems less than ideal.

AEFeinstein, 2025-02-19:
Maybe I'm misunderstanding the gist, but couldn't you have the CNFS blob creator also generate a large enum with all the file offsets like:
```c
typedef enum {
  CNFS_IMAGE_A_WSG = 0x0100,
  CNFS_IMAGE_A_WSG = 0x0200,
  CNFS_JINGLE_MID = 0x0700,
} cnfsOffset_t;
```
And then reference using that enum rather than a string filename?

dylwhich, 2025-02-19:
Hmm... yeah, I think that would be preferable, but could we actually do that? The enum constants would have to be in a `.h` file, and you can't just `extern` an enum. Or can you?? Or we could just generate another file for CNFS, though I think even that might confuse vscode until you do the asset processing at least once.

AEFeinstein, 2025-02-20:
You can't `extern` an enum because it's not a variable. It's more like a set of numerically unique `#defines`. It's very doable.

dylwhich, 2025-05-14:
Another potential enhancement to the asset preprocessor I've thought about a couple times is the ability to have file-specific settings that the preprocessor can use. The main use case I can think of here would be for images; right now there's no way to have a specific image be dithered since it's controlled by a `#define` in the processor code. Here's how it would work:

* We define some file extension, like `.opts` or `.assetconf` or even `.ini` or something
* When loading an input file like `image.png`, we also check for `image.png.opts` (or just `image.opts`?) in the same directory, and load that.
* The image processor can do whatever it wants with the contents of that options file, but it could just contain something like `dither=1`
* Aside from the single-file options, we could also just put a `.opts` file inside a directory and have that apply to everything within that directory. So, if you have a bunch of big background images that need to be dithered, you just put them all in a directory along with a `.opts` file that says `dither=1` and you're set!

How to do this technically? We'd just need to have an optional function for each asset preprocessor that takes the file contents and returns _some blob/struct_ that the processor recognizes, and have that passed in as a `void*`. Or we could implement a super simple key-value thing (maybe just use a hashmap?) that holds the settings so nobody has to bother with structs and casting.

AEFeinstein, 2025-05-16:
Oh I do like that. In the past I've either pre-dithered or pre-downsampled colors to make sure I got the desired outcome, but something like ini or yaml is more flexible. I'm also kinda doing this already with `.font.png`, but I'm pretty sure we don't want filenames to be basically a config string...

Definitely keep implementation simple. A non-hashed map should be fine. O(N) lookups won't kill anyone.

PR #447 (2025-05-24): Asset preprocessor improvements
### Description

This is a fairly extensive overhaul of the assets preprocessor, slightly beyond what's described in #400.
Main changes:
* `assets.conf` -- now there's a config file! you don't need to modify the C code if you want to use an existing processor with a new file extension
* `.opts` files -- you can use `<filename>.opts` to change how `<filename>.<ext>` is processed, or `dir/.opts` to affect everything inside `dir/`
* Exec processors -- you can define an asset processor that just runs a shell command now, which I've set up for sokoban as an example.  [... cut]
closed 2025-05-24 by dylwhich