AEFeinstein/Super-2024-Swadge-FW #411: RTC/ESP Low power mode

johnnywycliffe, 2025-01-27:
**Summary**
Something that has come up repeatedly is the desire for clock based games. I have a few ideas I'd like to try out, and there's a bunch more suggested in the slack. 

An RTC enables a constant clock once set, allowing modes to pass time while the swadge is on and keep the batteries alive for longer.

**Technical Spec**
There are two main proposed options:
1. The ESP's already-implemented low power mode. The 
2. An external RTC

Option 1:
- No extra hardware: Entirely software driven. We already have a screensaver thing so we could even repurpose that.
- Low power mode is reasonably performant- should last months on the AAs
- Requires the clock to be set any time the swadge loses power
- Swadge must be on at all times. If a user shuts it off or yanks a battery, the clock must be reset.

Option 2:
- Requires external power source. If we spring for a battery, will last a decade. A larger capacitor (TODO: Link slack message) will keep it running for a few hours while batteries are swapped out.
- Persists past taking out the AAs to replace them. in either case
- Should cost about $1 additional per swadge if we use a capcitor, high end. With a battery it would be closer to $2 per.

**UI**
A screen explaining about the RTC is case of option 1. Option 2 would not require this.

**How to Test**
Drop swadge into a low power mode/turn it off, then load it back up and see if a clock function has updated properly.

AEFeinstein, 2025-01-27:
Quick notes while I'm exhausted:
* Do we care about wall clock time, or just the passage of time? Guess it depends on the game. We may not care if it resets when batteries get swapped.
* I don't love keeping the ESP on the whole time. The only way to guarantee it stays on is by removing the power switch, but that's not feasiable.
* For option 2, I think the cost would be closer to $0.11, not a dollar. Here's an [RTC for $0.07 @ qty 2500](https://www.lcsc.com/product-detail/Real-Time-Clocks_XBLW-DS1307DTR-XBLW_C18723598.html) and here's a [crystal for $0.04 @ qty 1000](https://www.lcsc.com/product-detail/Crystals_CJKJ-C30732-768NCXE20_C19725234.html). 
* Option 2 doesn't require an external power source. We could use the AAs for the RTC, but route the trace _before_ the power switch such that the RTC is always on.

johnnywycliffe, 2025-01-27:
- I think giving the Swadge timekeeping abilities adds a slew of utilities. The ability to have an alarm clock, or just a spare clock is immediately useful. 
- I mean, I don't either. I personally want an external button cell, but that's the most expensive option, because of course it is. I think we could pre-install batteries and have a paper tab the user pulls instead of the power switch? The ESP draws very little power in its deep sleep mode.
- I refuse to actually price this out. I'm sure you're correct too, I'm just being stubborn. I'll update the topic accordingly.
- Leaching off the AAs still fails the replacement test. I'd prefer the capacitor over that, since it would persist past battery replacement, which is the entire point (in my mind) of having an external RTC over the built in one.

cnlohr, 2025-01-27:
I have used the RTC processors within the esp32-s2, and they are quite competent (though annoyingly slow for anything active).  That said, it is possible.

The power tree is a little annoying, in that it's hard to get the processor to go like super low power, but it can be done reasonably well. I am happy to do power budget testing in the coming weeks.  

[image]

johnnywycliffe, 2025-04-26:
https://github.com/AEFeinstein/Super-2026-Swadge-HW/issues/2

johnnywycliffe, 2025-05-31:
Since we've basically decided on a low power mode method (swadgepass + light dances) can I write in a timekeeper? Something that just counts minutes swadge has been turned on?

I would use this for the cozy camper game I wanted to make

johnnywycliffe, 2025-05-31:
I was looking for each minute. As you mentioned, about 6k writes over the course of the event.

johnnywycliffe, 2025-05-31:
I think we could make it so it can be disabled whenever. Maybe only enable it if the cozy camper mode is enabled.

I will mention that the ESP has automatic wear leveling.

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/storage/nvs_flash.html

AEFeinstein, 2025-05-31:
Wear leveling is nice, but still finite. Entropy comes for us all. I didn't think that Cozy Camper was going to be an enable/disable thing. I could understand not counting time until CC is used for the first time I guess?

johnnywycliffe, 2025-05-31:
Basically, just holding off the doom until it's explicitly asked for.

I suppose I can put it at 5 minutes, that would extend it a bit.

jarettmillard, 2025-05-31:
You could count time without writing to NVS while Cozy Camper is active, but rely on the 5-minute writes when it's not. Most idle games tend to have increasing times for progression as the game advances.

johnnywycliffe, 2025-05-31:
Yeah. I just want to try to avoid losing a ton of progress if you randomly shut off your device. If it's every five minutes, then that means playing a quick few microgames results in not even a single write. It's a problem of average play time being below the write time.

AEFeinstein, 2025-05-31:
Right, if there was a way to detect shutdown and write the time once then, that'd be perfect, but there isn't.

johnnywycliffe, 2025-10-25:
Closing, as swadgepass effectively is the answer to this issue.

closed 2025-10-25 by johnnywycliffe