AEFeinstein/Super-2024-Swadge-FW #417: Swadge Avatar Creator

emilyanthony4244, 2025-01-31:
An Avatar/dress up game to create a lil guy to live in your swadge.

Inspired by the classic flash dress up games in browser, Style Savvy from the 3DS, etc.

Examples:
https://www.dolldivine.com/classic-flash-games
https://www.avatarsinpixels.com/minipix/clothing/Body
Bitmoji
etc

This could be used to be fed into an avatar for SwadgePass.

Examples at 200px height for the game and 36px for the avatar to be fed to SwadgePass:

[image]

[image]

AEFeinstein, 2025-02-06:
All for it. I think the minimum for a Mii-like would be face only:
* Head Shape
* Hair
    * Style
    * Color
* Eyes
    * Style
    * Color
* Nose
* Mouth
    * Beard?
* Ears

For the hair and eyes, we may need to revisit [palette swapping](https://adam.feinste.in/Super-2024-Swadge-FW/wsgPalette_8h.html). Perhaps those elements can be drawn in grayscale so that each palette could be a tidy six colors instead?

AEFeinstein, 2025-03-28:
@LittleLunaToon now would be a really good time to start thinking about (and doodling) what face-parts look like. If each face is in a 36x36 canvas, how large would each face-part be, roughly? Eyes/nose/mouth/ears/hair.

Miis got away with solid color clothes, which isn't much of a dress-up, but I do like that aesthetic and it keeps things simpler. @emilyanthony4244, vibes?

AEFeinstein, 2025-03-28:
Oh, I see some stuff in https://github.com/AEFeinstein/Super-2024-Swadge-FW/tree/swadgesona. Neat. Do we prefer Swadgesona or Swavatar?

AEFeinstein, 2025-05-21:
The user survey should be part of the avatar creator as well.