AEFeinstein/Super-2024-Swadge-FW #421: WebAssembly Support

dylwhich, 2025-02-14:
## Summary

We should be able to run the Swadge Emulator in WebAssembly!

**Why?**
1. Reduced barrier to entry -- If WASM is supported, it would be possible for anyone with a GitHub account to go to https://github.dev/AEFeinstein/Super-2024-Swadge-FW/ and work on the Swadge without ever having to set up an environment on their local machine.
2. Easier to get mode feedback -- It's possible to [expose codespace ports publicly](https://docs.github.com/en/codespaces/developing-in-a-codespace/forwarding-ports-in-your-codespace#sharing-a-port), which means you can let someone try out a game you're actively developing by just sharing a link. No need to even commit, you can try out changes almost in real time!
3. It would be really neat!

## Technical Spec

Functionally this is pretty similar to the [SAO wasm emulator](https://github.com/cnlohr/Super-2024-Swadge-FW/tree/ch32v003programmer/tools/sandbox_test/test_programmer/sao047b/web). A few parts of the emulator implementation would need to be modified to support WebAssembly, in particular anything that does file I/O will need an implementation that uses some browser API instead. We could do the implementation in two stages, with the first stage being a minimum-viable implementation that renders to the screen and takes keyboard input, but doesn't support audio, saving data, etc. The second stage would be a more complete implementation with audio, NVS support, and potentially joystick/MIDI support, replay support, screenshots, and other goodies.

Minimum prereqs for "it works":
* [x] WASM-compatible rendering and key/mouse input (rawdraw)
* [ ] stdlibc implementation suitable for WASM
* [x] WASM build process for emulator

Stage 2:
* [x] alternative NVS implementation (LocalStorage)
* [ ] alternative joystick implementation (gamepad API)
* [ ] alternative MIDI implementation (WebMIDI)
* [ ] advanced emulator features that use files (replay, screenshots, screen recordings, opening MIDIs)
* [ ] pass args via URL, e.g. `https://example.com/?mode=Touch+Test` would be equivalent to `swadge_emulator --mode "Touch Test"`

## UI

## Mockups

## How to Test

AEFeinstein, 2025-02-14:
This is really cool and I'm mostly going to lean back and let you have at it. Just make sure any non-C is pretty well commented, because it's not my native tongue. I mean, any C should be commented to, natch.

dylwhich, 2025-02-16:
This is the latest weblibc I could find: https://github.com/cntools/rawdraw/tree/weblibc/wasm/weblibc

It's not 100% complete though and based on a pretty old rawdraw, so I [forked it](https://github.com/dylwhich/rawdraw/tree/weblibc) and rebased it onto the latest so I can use it as a submodule here in the [wasm-2 branch](https://github.com/AEFeinstein/Super-2024-Swadge-FW/compare/main...wasm-2).

I'm working through implementing the missing functions and identifiers myself, here's everything that's missing in the current firmware:

Functions:
* [ ] strdup
* [ ] nanosleep
* [ ] strtof
* [ ] rand
* [ ] strcasecmp
* [ ] fgetc
* [ ] strncacescmp
* [ ] srand
* [ ] usleep
* [ ] clock_gettime
* [ ] strtok
* [ ] strpbrk

And also these, but they can all be directly exported from javascript so they just need a declaration really:
* [ ] cosf
* [ ] sinf
* [ ] abs
* [ ] atan2
* [ ] sqrt
* [ ] fabsf
* [ ] sqrtf
* [ ] floor
* [ ] sin
* [ ] cos
* [ ] labs
* [ ] atan2f
* [ ] log
* [ ] pow

Defines / enum values
* [ ] M_PI
* [ ] CLOCK_MONOTONIC
* [x] PRI... and  from inttypes.h

Structs:
* [ ] struct timespec

Files:
* [x] sys/stat.h
* [x] inttypes.h
* [ ] dirent.h

Other things:
* [ ] assert() is missing some parentheses apparently
* [ ] my PRIuPTR/64 size modifiers are wrong (wasm64 defines backwards?)

AEFeinstein, 2025-02-16:
Knowing nearly nothing about wasm, I'm very surprised weblibc isn't a thing
that already wholly exists

cnlohr, 2025-04-21:
I have been working more on this recently and have several other functions ready, along with some tweaks in my bad apple project.