AEFeinstein/Super-2024-Swadge-FW #422: Windows

johnnywycliffe, 2025-02-28:
## Description
On windows, the asset preprocessor seems to break builds. Specifically, lines 342-345 of the makefile. It appears to be something to do with the CNFS system. All submodules are present AFAIK and updated.

Tested on 2 computers, both logs included.

## Expected behavior
Swadge FW/emulator builds.

## Reproduction Steps
On Windows: Make a new `esp/` folder, install the IDF, and clone the git repo. Use `export.ps1`, then `idf.py build`. It should fail in the asset preprocessor phase.

## Logs
`idf.py build`
```
[1/1116] Generating project_elf_src_esp32s2.c
[2/1116] Generating always_rebuild

[3/1116] Generating C:/Users/Johnn/esp/Super-2024-Swadge-FW/build/esp-idf/esp_system/ld/memory.ld linker script...
[4/1116] Generating C:/Users/Johnn/esp/Super-2024-Swadge-FW/build/esp-idf/esp_system/ld/sections.ld.in linker script...
[5/1116] Building ASM object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/xtensa_context.S.obj
[6/1116] Building ASM object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/xtensa_intr_asm.S.obj
[7/1116] Building ASM object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/xtensa_vectors.S.obj
[8/1116] Building C object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/eri.c.obj
[9/1116] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,4032K,

*******************************************************************************
[10/1116] Building C object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/xt_trax.c.obj
[11/1116] Building C object esp-idf/xtensa/CMakeFiles/__idf_xtensa.dir/xtensa_intr.c.obj
[12/1116] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32s2/esp_efuse_table.c.obj
[13/1116] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32s2/esp_efuse_rtc_calib.c.obj
[14/1116] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32s2/esp_efuse_fields.c.obj
[15/1116] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32s2/esp_efuse_utility.c.obj
[16/1116] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/esp_efuse_fields.c.obj
[17/1116] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/esp32s2/esp_efuse_rtc_table.c.obj
[18/1116] Building C object esp-idf/esp_mm/CMakeFiles/__idf_esp_mm.dir/port/esp32s2/ext_mem_layout.c.obj
[19/1116] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/esp_efuse_api.c.obj
[20/1116] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/esp_efuse_utility.c.obj
[21/1116] Building C object esp-idf/efuse/CMakeFiles/__idf_efuse.dir/src/efuse_controller/keys/with_key_purposes/esp_efuse_api_key.c.obj
[22/1116] Building C object esp-idf/esp_mm/CMakeFiles/__idf_esp_mm.dir/esp_cache.c.obj
[23/1116] Building C object esp-idf/esp_ringbuf/CMakeFiles/__idf_esp_ringbuf.dir/ringbuf.c.obj
[24/1116] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/gpio/gpio_glitch_filter_ops.c.obj
[25/1116] Generating C:/Users/Johnn/esp/Super-2024-Swadge-FW/main/utils/cnfs_image.c, C:/Users/Johnn/esp/Super-2024-Swadge-FW/main/utils/cnfs_image.h
FAILED: C:/Users/Johnn/esp/Super-2024-Swadge-FW/main/utils/cnfs_image.c C:/Users/Johnn/esp/Super-2024-Swadge-FW/main/utils/cnfs_image.h 
cmd.exe /C "cd /D C:\Users\Johnn\esp\Super-2024-Swadge-FW\build\esp-idf\main && python C:/Users/Johnn/esp/Super-2024-Swadge-FW/main/../tools/soko/soko_tmx_preprocessor.py C:/Users/Johnn/esp/Super-2024-Swadge-FW/main/../assets/soko/ C:/Users/Johnn/esp/Super-2024-Swadge-FW/main/../assets_image/ && make -C C:/Users/Johnn/esp/Super-2024-Swadge-FW/main/../tools/assets_preprocessor/ && C:\Users\Johnn\esp\Super-2024-Swadge-FW\main\..\tools\assets_preprocessor\assets_preprocessor -i C:/Users/Johnn/esp/Super-2024-Swadge-FW/main/../assets/ -o C:/Users/Johnn/esp/Super-2024-Swadge-FW/main/../assets_image/ && make -C C:/Users/Johnn/esp/Super-2024-Swadge-FW/main/../tools/cnfs && C:\Users\Johnn\esp\Super-2024-Swadge-FW\main\..\tools\cnfs\cnfs_gen C:/Users/Johnn/esp/Super-2024-Swadge-FW/main/../assets_image/ C:/Users/Johnn/esp/Super-2024-Swadge-FW/main/utils/cnfs_image.c C:/Users/Johnn/esp/Super-2024-Swadge-FW/main/utils/cnfs_image.h"
Starting soko tmx conversion
Completed soko tmx converstion. 0/31 tmx files converted. 0/0 - 0 (of converted) bytes saved with compression.
make: Entering directory '/c/Users/Johnn/esp/Super-2024-Swadge-FW/tools/assets_preprocessor'
FIND: Parameter format not correct
File not found - *.[c]
FIND: Parameter format not correct
File not found - *.[c]
gcc  obj/../../emulator/src/idf/esp_heap_caps.o   -lm -static-libstdc++ -ggdb -o assets_preprocessor
/usr/lib/gcc/x86_64-pc-msys/13.2.0/../../../../x86_64-pc-msys/bin/ld: /usr/lib/gcc/x86_64-pc-msys/13.2.0/../../../libmsys-2.0.a(libcmain.o):(.text.startup+0x7f): undefined reference to `WinMain'
collect2: error: ld returned 1 exit status
make: *** [makefile:145: assets_preprocessor] Error 1
make: Leaving directory '/c/Users/Johnn/esp/Super-2024-Swadge-FW/tools/assets_preprocessor'
[26/1116] Building C object esp-idf/esp_mm/CMakeFiles/__idf_esp_mm.dir/esp_mmu_map.c.obj
[27/1116] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/dac/dac_oneshot.c.obj
[28/1116] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/spi/spi_bus_lock.c.obj
[29/1116] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/gpio/rtc_io.c.obj
[30/1116] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/dac/dac_common.c.obj
[31/1116] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/dac/dac_cosine.c.obj
[32/1116] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/deprecated/dac_common_legacy.c.obj
[... 73 lines cut ...]
make[1]: Entering directory '/c/Users/Johnn/esp/Super-2024-Swadge-FW/tools/assets_preprocessor'
FIND: Parameter format not correct
File not found - *.[c]
FIND: Parameter format not correct
FIND: Parameter format not correct
gcc -g -std=c99 -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-error=unused-but-set-variable -Wno-old-style-declaration -Wno-missing-field-initializers -Wundef -Wformat=2 -Winvalid-pch -Wlogical-op -Wmissing-format-attribute -Wmissing-include-dirs -Wpointer-arith -Wunused-local-typedefs -Wuninitialized -Wshadow -Wredundant-decls -Wjump-misses-init -Wswitch-enum -Wcast-align -Wformat-nonliteral -Wno-switch-default -Wunused -Wunused-macros -Wmissing-declarations -Wmissing-prototypes -Wcast-qual -Wno-switch    -I../../emulator/idf-inc/ -c ../../emulator/src/idf/esp_heap_caps.c -o obj/../../emulator/src/idf/esp_heap_caps.o
FIND: Parameter format not correct
File not found - *.[c]
gcc  obj/../../emulator/src/idf/esp_heap_caps.o   -lm -static-libstdc++ -ggdb -o assets_preprocessor
/usr/lib/gcc/x86_64-pc-msys/13.2.0/../../../../x86_64-pc-msys/bin/ld: /usr/lib/gcc/x86_64-pc-msys/13.2.0/../../../libmsys-2.0.a(libcmain.o):(.text.startup+0x7f): undefined reference to `WinMain'
collect2: error: ld returned 1 exit status
make[1]: *** [makefile:145: assets_preprocessor] Error 1
make[1]: Leaving directory '/c/Users/Johnn/esp/Super-2024-Swadge-FW/tools/assets_preprocessor'
make: *** [makefile:342: main/utils/cnfs_image.c] Error 2
```

## Affected Version
 - Platform: Windows
 - git branch: Main
 - git hash: [c8bf3d6](https://github.com/AEFeinstein/Super-2024-Swadge-FW/commit/c8bf3d69d13a0c93d36d8656b436a6159e13ea03)

## Additional Notes
Tested on two machines: Luna's and mine. We've both encountered the same issues, and I have copied her log to this issue as well.

Both machines were fresh installs following the "configuring a development environment" page of the wiki, and both copied and pasted the commands, so there was no forgetting to copy over submodules.

AEFeinstein, 2025-02-28:
Other data point, CI is still building ok 
https://github.com/AEFeinstein/Super-2024-Swadge-FW/actions/runs/13198751061/job/36845776142

I need to strip out all the games and stuff for next year, maybe the problem will disappear then...

AEFeinstein, 2025-02-28:
Is the windows path set correctly to use the GNU find rather than MSFT find?

johnnywycliffe, 2025-02-28:
It appears to be using the microsoft find commend, since /? loads help and --version fails. 

How do I set it correctly?

johnnywycliffe, 2025-02-28:
Checking the makefile, I found where it links the find command. I tried replacing the command with the direct msys path to the file and it still failed.

```
FIND:=find
ifeq ($(HOST_OS),Windows)
	FIND:=$(shell cygpath C:\msys64\usr\bin\find.exe | grep bin | grep -v " ")
endif
```

AEFeinstein, 2025-02-28:
Step 6 here
https://github.com/AEFeinstein/Super-2024-Swadge-FW/blob/main/docs/SETUP.md#configuring-a-windows-environment

johnnywycliffe, 2025-02-28:
Yes, the path on both computers is set up correctly. Here's mine for reference:

[image]

`make` wouldn't work if they weren't set up. That was an issue I had in the past.

dylwhich, 2025-03-01:
> Checking the makefile, I found where it links the find command. I tried replacing the command with the direct msys path to the file and it still failed.
> 
> ```
> FIND:=find
> ifeq ($(HOST_OS),Windows)
> 	FIND:=$(shell cygpath C:\msys64\usr\bin\find.exe | grep bin | grep -v " ")
> endif
> ```

That's in the top-level makefile, but I suspect this doesn't pass through to the `tools/assets_preprocessor` makefile, which just has this:
```
FIND = find
```

dylwhich, 2025-03-01:
Huh, by some bizarre coincidence the latest `main` has a completely unrelated compile error on Linux that also somehow doesn't affect the CI build.

johnnywycliffe, 2025-03-02:
Closed, error was with windows path.

PR #424 (2025-03-03): Use proper find in windows for tools/asset_preprocessor. Fixes #422
### Description

Just copies the windows-specific find from makefile into tools/assets_preprocessor/makefile

### Test Instructions

@johnnywycliffe should try to build this branch and see if it fixes #422. And the CI build will make sure this doesn't break things for anyone else.

### Ticket Links

Fixes #422 

### Readiness Checklist
- [x] I have run `make format` to format the changes
- [x] I have compiled the firmware and the changes have no warnings
- [x] I have compiled the emulator and the changes have no warnings
- [x] [... cut]
commit d318b740 (2025-03-03): Use proper find in windows for tools/asset_preprocessor. Fixes #422
closed 2025-03-02 by johnnywycliffe