AEFeinstein/Super-2024-Swadge-FW #435: Swadgedoku

dylwhich, 2025-04-09:
## Summary
You know sudoku? It's literally sudoku. Maybe with a gimmick or two if I can come up with a good one. Maybe some form of (cosmetic?) unlockables / power-ups?

It should have infinite random puzzles, either with a selectable difficulty or with sets of curated levels of several difficulty levels. It should also have an analytical solver / explainer built in, which can give hints for how to progress and explain the logic used.

It would also be cool to support sudoku variants or alternate bases. You can do binary sudoku (there are only like 14 unique games, all of them trivial, but would be funny if we have unlockables) And there's "super sudoku" where it's 5 normal sudoku grids arranged in an X, where the 4 corner (3x3) boxes are shared between the center and corner sudoku grid. 

## Technical Spec

Well, it's sudoku. There will be a 2D array representing the grid with numeric values (`uint8_t`), 9x9 or possibly 2x2 or 16x16. 0 represents an un-filled value. There could also be 3 arrays of bit-fields representing the used values for each row, column, and box, to make hinting/solving logic faster and more convenient.

For drawing the grid, I can probably yoink a lot of the UTTT code. And also maybe for drawing the "explain mode" lines and such.

## UI

## Mockups

## How to Test

AEFeinstein, 2025-04-09:
Ultimate sudoku, with 729 cells total. Everyone will love it.

dylwhich, 2025-05-14:
A couple other thoughts here about generating levels; I think it would be a good start to have a set of predefined curated puzzles that ramp up in difficulty, as a sort of "campaign mode" or an extended tutorial. New techniques would be required to solve puzzles as the player progresses, and after completing the last (or one of the last) levels you should have demonstrated all the solving techniques needed to solve the hardest puzzles we can generate, assuming that we can do level generation, because...

Level generation is pretty hard! You need to have a good solver in order to do this, and then the algorithm isn't _hard_ but it's not that simple either. Most of the literature I found involves backtracking as you generate the level, so it's something like this:
* Advance to the next cell
* Place a random valid number in the cell
* Check that no constraints have been violated. If so, pick a different number and try again. If all possibilities have been tried, backtrack further and continue.
* You now have a complete, solved sudoku. Now to create the puzzle
* Select a random square and delete its value
* Check that the puzzle still has a unique solution. If not, select a different square. Record the simplest strategy needed to figure out the deleted number back (for determining difficulty).
* Continue until the puzzle is sufficiently difficult

So, difficulty selection is likely to be the biggest can of worms here. I couldn't find much literature on how to generate puzzles of a particular difficulty. Most people seemed to suggest "just keep generating puzzles until you find one with the desired difficulty" but I don't even know how long the ESP will take to generate a single puzzle so I'm hoping that's not the only way. I don't know if this would actually work in practice, but I think the difficulty could be "steered" a bit at the "delete random numbers" step. Basically each difficulty gets a budget of certain techniques (so e.g. Beginner difficulty only get the most basic strategies and nothing else). Then, if possible, determine which numbers may be removed with only the allowed strategies, and randomly select one of those. Or, if that's not possible (or too hard), just use backtracking whenever an illegal technique is needed to continue solving.

If it turns out we can't manage fully randomized sudokus on the swadge, there's another way to do generation that sounds a bit less intensive. You can start with a valid sudoku (so we could generate a bunch off-swadge) and then randomly permute it by swapping around rows/columns within boxes and swapping full sets of 3 rows/columns within a box, and by swapping digit values. Allegedly this preserves the solvability and difficulty while making the puzzle 'different enough' but imo it's a bit cheaty and if it's at all possible I'd like to be able to fully generate random puzzles from scratch instead.

AEFeinstein, 2025-05-16:
> You can't put checkboxes in an ordered list

No, but if you put a multiple of four spaces before a line it'll indent and nest under a nicely
- [ ] like
    - [ ] this
    
Anyway, that's a hearty list of stuff to implement. I do like the curated levels that get harder. I wonder if people will be generally more familiar with Sudoku than Nonograms, and if that helps at all.

Level generation does sound strangely fiendish, though all the best puzzles are best because they're not easily solved by CPUs.

> Check that the puzzle still has a unique solution. If not, select a different square.

Is it really a requirement to have a single unique solution? What's the harm in having a puzzle that's solveable multiple ways?

> You can start with a valid sudoku (so we could generate a bunch off-swadge) and then randomly permute it

I swear I was thinking about this while reading the prior paragraph. I was trying to figure out how to permute cells validly and hadn't gotten to the blocks / rows / cols yet. This feels doable, and it's not cheaty if the puzzles all feel unique!

closed 2025-12-29 by AEFeinstein