AEFeinstein/Super-2024-Swadge-FW #437: Trophy System - Update skety queue to linkedlist

johnnywycliffe, 2025-04-26:
## Summary
Trophy system currently uses a array to hold the trophy objects and manage their data. I'd like to have it moved to a linked list, but:
- I haven't used the Swadge Linkedlist before
- I've been stupid busy
- I can't figure out how not to leak memory given how it's constructed.

THis needs to be done in the 2026/trophy branch, I'm still working on it, but I don't have the time to figure this out properly.

## Technical Spec
In trophy.c:
```C
trophyDataWrapper_t trophyQueue[TROPHY_MAX_BANNERS]; //< List of trophy updates to display
```
should be replaced by a linked list.

The mode should automatically check for another queued trophy announcement before shutting the animator down for a refresh later when it reaches the end, regardless of if the trophy display mode is set to sliding or just displaying the notification.

## How to Test
- Load trophy test mode
- Clear trophies for clean slate (Start->clear data)
- Spam "B" button and observe trophies. Should be ten notifications in sequence. No duplicates, nothing overwritten, nothing missing.

This issue is part of #397

AEFeinstein, 2025-05-05:
@johnnywycliffe I followed the test instructions and it _kinda_ works, except the banners are only shown for a faction of a second. This seems to be consistent with the branch before my changes?

johnnywycliffe, 2025-05-05:
I set the rate to be pretty fast so I could test ten messages in a row quickly without having to wait 30 seconds. 

I'll check to make sure it's not actually broken when I get home

AEFeinstein, 2025-05-05:
Sounds good, I'll be on standby

AEFeinstein, 2025-05-09:
OK, I think it's all cleared up in four commits
* c3157e64dcb6161d2f09bcc66c33cc2b1dbb6932 - Fixed timing being off by an order of magnitude, also fixed some warnings
* 115ae15c5fa6676ddc7aaede4d8747bfc5f17dd4 - Fix typos, because I'm like that
* 13da8f7cd477cd2455dc8ceff3b2ab5a786cdbd4 - Started by fixing the use-after-free, ended up rewriting animation. File is now 100 lines shorter and works nicely
* 2695f227a83f237ac27d699e40b129f5a79180e1 - The refactor ignored timings in `trophySettings_t`, so this uses them again

johnnywycliffe, 2025-05-10:
Closed - Resolved in 5cbf923

closed 2025-05-10 by johnnywycliffe