AEFeinstein/Super-2024-Swadge-FW #438: Dice Roller

johnnywycliffe, 2025-04-26:
## Summary
This issue is set up to track dice roller discussion.

I will update as required.

## Technical Spec
- Accelerometer: Used to shake dice.

AEFeinstein, 2025-04-26:
Note that we had a dice roller on the Squarewavebird. That was before we started moving old modes to the `attic`, but you can find it here: https://github.com/AEFeinstein/Super-2023-Swadge-FW/blob/main/main/modes/mode_diceroller.c. 

@dylwhich also has a branch https://github.com/AEFeinstein/Super-2024-Swadge-FW/tree/2026/dice-roller-thing. I don't know much about it.

timdev-dash, 2025-04-29:
If possible, it would also be nice to have the option for the roller to provide the total results of the roll. Math is hard with math rocks!

Eg: Rolling 8d6. Results are 6,4,3,5,1,1,3,2
Total: 25

AEFeinstein, 2025-05-06:
@timdev-dash yep, the dice roller from a few years ago had totals:
[image]

AEFeinstein, 2025-05-08:
@johnnywycliffe The port and refresh does everything you ask for, except for 3D and palette swaps. I'm hesitant to get too artistic before the theme is public, since we may want to apply some of that good stuff.

If you want to rewrite the render bits in 3D, that's on you.