AEFeinstein/Super-2024-Swadge-FW #439: Pedometer mode

johnnywycliffe, 2025-04-27:
## Summary
Pedometer. MAG already has a steps competition, now to weaponize it!

## Technical Spec
- Need to be able to put swadge into low power mode
- Needs to still be able to access the accelerometer.
- Needs some sort of sway counter to count steps

## UI
Menu:
- Reset steps
- Trip 1
- Trip 2
- Turn off screen

## How to Test
Put on swadge and walk around for a bit. Make sure it's roughly keeping up with steps.

AEFeinstein, 2025-04-27:
I believe the IMU we have has some sort of step counting. I don't have the datasheet in front of me

AEFeinstein, 2025-05-05:
[LSM6DS3TR datasheet](https://datasheet.lcsc.com/lcsc/2102261805_STMicroelectronics-LSM6DS3TR-C_C967633.pdf). 

`CTRL10_C` has `PEDO_EN` and `FUNC_EN` for enabling. There's probably other config and setup to do.

`STEP_TIMESTAMP_H`, `STEP_TIMESTAMP_L`, `STEP_COUNTER_H`, `STEP_COUNTER_L` seem promising for reading values.

Unrelated, I see that it has a thermometer too. Neat.

johnnywycliffe, 2025-05-05:
*We must dual wield the thermometers for redundancy*