AEFeinstein/Super-2024-Swadge-FW #440: PvP Mini Chess

DebrisHauler, 2025-05-03:
## Summary
Mini Chess (working title) is a turn-based PvP game on a 5x5 isometric board.
Win by moving your Band Leader in front of your opponent's Studio Mixer to play your music.
Or win by destroying your opponent's Band Leader.
The Game Over screen features a still graphic of the winning Band Leader playing their actual midi music.

## Inspiration
Inspired by Onitama, but nearly everything about it is redesigned with different mechanics and rules. The pixel art style is inspired by a game that shall not be mentioned until MAGFest announces their theme.

## Thoughts Behind the Design
In Mini Chess, it is unique how players create the cards (Albums) over the course game and manage their hand size of zero to four which also serves in a way as health to be discarded. There is also a shared pool jokingly called the Creative Commons which _should_ regulate the balance of the gameplay and engage players in influencing one another. The pool can even generate cards when it is empty. A player can refrain from taking an action on the board in order to draw from the pool. And when an attack lands successfully, that card is automatically sent to the pool, and they take a different card from the pool. Also, when being attacked, a player may send cards to the pool instead of sacrificing a unit.

## Start
On the back rows similar to chess, players are each set up with 2 Groupies, 1 Band Leader, and 2 more Groupies all in a row. The Studio Mixers are accessible at the starting locations of the two Band Leaders.
Players start with one album each. Albums are never hidden. A Starter Album contains only a random Interval with blue color. More on Intervals and Colors in the next section.
The Creative Commons also holds a random Starter Album.
Players can see their opponent's Albums and the Creative Commons throughout the game by pressing the pause button (4x3 UI layout).

## Albums
Albums are made of different colored Intervals. An Interval refers to a relative position from the acting unit, such as "left 2 and back 1". The color of an Interval can be: 
- Blue: move
- Red: attack (ranged)
- Purple: move & attack. Animated by leaping or warping to the purple square and attacking at the purple square. Return to the starting position IF the opponent burns two Albums.

## Attack
If a player attacks an opponent’s unit, the Album used for the attack goes to the Creative Commons, and the aggressor picks a different Album from the Creative Commons.
The victim of the attack chooses to send two of their Albums to the Creative Commons OR sacrifice their unit. The victim’s unit must be destroyed if the player has less than two Albums in hand. It would be funny to depict as “somebody left the band”.

## Player Turn
**Phase 1:**
The player first optionally improves an Album that is already in hand.
When improving an Album, a player can:
Add an interval with a random color, OR
Add a color at a random interval.
**Phase 2:**
They choose to take an album from the Creative Commons OR they use one of their Albums.
Note: When the Creative Commons becomes empty, it is replenished with a random (Starter?) Album.
Using an Album: The player picks a unit (one of their Groupies or Band Leader) to use one of the Intervals in one of their Albums. A unit cannot move or attack off the board, or into allies. In this case, the only choice is to draw from the Creative Commons.

## Swadge Pass Integration
Not married to any ideas yet:
- Visitors make the game a little crazier for both sides by adding pseudo random yet symmetrical board spots that grant either player an extra Album when they move there. If the pseudo random spots are all filled from several swadge passes, then additional guests grant both players extra starting Albums.
- Have a swadge pass share a user-made midi file for the end of the game.
- Swavatar guests dance to the midi music in a little concert.
- Swavatar guests cheer you on during battle just above the board.

## Nice To Have
Choose playable bands from the festival's theme. Unlock them via achievements from other swadge games. The game could be quite art heavy with all the Characters & Groupies animated in isometric forwards and backwards for Idle, Range Attack, Melee Attack, Step Back to Share a Tile with a Melee Attack, and Destroyed.

AEFeinstein, 2025-05-04:
> When improving an Album, a player can:
> Add an interval with a random color, OR
> Add a color at a random interval.

What happens when an Album has two blue (move) intervals?

DebrisHauler, 2025-05-04:
You get to pick which square to move to when you use that Album. It may be a little confusing until we can see it on screen. But an Album could be loaded up with many Intervals. And you'd pick one of the Intervals to use.

DebrisHauler, 2025-05-06:
Might sound better name them Tracks instead of Intervals. So you add Tracks to an Album. "Track" somewhat evokes a sense of going somewhere as well.

AEFeinstein, 2025-05-06:
Oh I like Tracks a lot more